fdivitto/FabGL #1: bitmaps.h

techno94, 2019-02-24:
Hello,
it's an awesome librarie
i don't find how to generate the bitmaps.h
thanks a lot
[image]

techno94, 2019-02-24:
and how to make the vtanimations.h
thanks a lot

fdivitto, 2019-02-26:
Hello,
I just added img1sprite tool. Please look at:

https://github.com/fdivitto/FabGL/tree/master/tools/img2bitmap <https://github.com/fdivitto/FabGL/tree/master/tools/img2bitmap>

fdivitto, 2019-02-26:
The content of vtanimations.h is just an ASCII file (ANSI/VT100 codes) converted to C structure. You can find a lot of these files on the net (sometime with ASC extension).
To convert the to C structure just execute (on Linux or Mac):

xxd --include inputfile.asc

closed 2019-02-26 by fdivitto