fdivitto/FabGL #9: New modeline qualifier similar to `DoubleScan`,  `ScanLine` to output a blank scanline on every second scanline.

dmcnaugh, 2019-04-03:
**Is your feature request related to a problem? Please describe.**
*A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]*
No

**Describe the solution you'd like**
*A clear and concise description of what you want to happen.*
I would like a new modeline qualifier similar to `DoubleScan`, maybe `ScanLine`, that instead of outputting a second copy of each scanline outputs a blank scanline on every second scanline. The aim is to produce a "retro" looking scanline effect

**Describe alternatives you've considered**
*A clear and concise description of any alternative solutions or features you've considered.*
Currently I use `DoubleScan` modelines, but they don't give the "retro" like I want.

**Additional context**
*Add any other context or screenshots about the feature request here.*
Looking through your i2s code I assume this can be done as this appears to be how you render scanlines outside the vertical viewport. I have tried my own modifications but I just don't understand the i2s code well enough to make it work.

fdivitto, 2019-04-05:
I've just added "MultiScanBlank" modeline option to do that. Anyway results are not so good when the background is not black. I'm sorry!
[image]
[image]

closed 2019-04-05 by fdivitto

dmcnaugh, 2019-04-06:
Grazie. Questo è eccellente.
Sei un genio. La tua soluzione è migliore della mia idea.
Grazie ancora.

fdivitto, 2019-04-06:
Grazie a te!