fdivitto/FabGL #10: Can't inherit from TerminalClass successfully

dmcnaugh, 2019-04-09:
I would like to inherit from TerminalClass to create a specific implementation that extends the base functionality you provide.
e.g.
```
class Vt100Class : public fabgl::TerminalClass {

};
Vt100Class Vt100;
```

However when I do this now I get runtime errors that I traced back to following member functions (methods) 

- TerminalClass::reset()
- TerminalClass::blinkTimerFunc(TimerHandle_t xTimer)
- TerminalClass::consumeInputQueue()
- TerminalClass::keyboardReaderTask(void * pvParameters)

Each of these member functions make direct references to the `Terminal.` object and this conflicts with my Vt100 object and makes the TerminalClass behave in an OO unfriendly way.

Can these member functions be be modified to make TerminalClass OO friendly or do they need to work this way?

fdivitto, 2019-04-09:
You're right, I'm fixing...

fdivitto, 2019-04-09:
Fixed, but now you have to create an instance of TerminalClass, as in the examples.

closed 2019-04-09 by fdivitto