fdivitto/FabGL #14: request for a simple draw demo

YordanYanakiev, 2019-05-30:
If it is possible I would like to have a simple demo regarding just drawing a few random dots or lines in the loop.

The point is to clearly understand the way it is supposed to be done this.
As simplest as possible will do.

bytefresser, 2019-05-30:
You can see it in example simpleterminalout. Check demo4.

YordanYanakiev, 2019-05-30:
I am not sure where is this "demo4" file. Could You provide a link please?

karlTH, 2019-05-30:
in this exemple
https://github.com/fdivitto/FabGL/blob/master/examples/SimpleTerminalOut/SimpleTerminalOut.ino

YordanYanakiev, 2019-05-30:
tried it, as well other demos, none seem to be working on my setup due some reason unknown:
```

#include "fabgl.h"

#if USE_8_COLORS
#define VGA_RED    GPIO_NUM_17
#define VGA_GREEN  GPIO_NUM_16
#define VGA_BLUE   GPIO_NUM_4
#define VGA_HSYNC  GPIO_NUM_15
#define VGA_VSYNC  GPIO_NUM_2
#elif USE_64_COLORS
#define VGA_RED1   GPIO_NUM_22
#define VGA_RED0   GPIO_NUM_21
#define VGA_GREEN1 GPIO_NUM_19
#define VGA_GREEN0 GPIO_NUM_18
#define VGA_BLUE1  GPIO_NUM_5
#define VGA_BLUE0  GPIO_NUM_4
#define VGA_HSYNC  GPIO_NUM_23
#define VGA_VSYNC  GPIO_NUM_15
#endif

// the setup function runs once when you press reset or power the board
void setup() 
{
	Serial.begin( 115200 ); delay( 500 ); Serial.write( "\n\n\n" ); // DEBUG ONLY

#if USE_8_COLORS
	VGAController.begin( VGA_RED, VGA_GREEN, VGA_BLUE, VGA_HSYNC, VGA_VSYNC );
#elif USE_64_COLORS
	VGAController.begin( VGA_RED1, VGA_RED0, VGA_GREEN1, VGA_GREEN0, VGA_BLUE1, VGA_BLUE0, VGA_HSYNC, VGA_VSYNC );
#endif

	VGAController.setResolution( VGA_640x400_70Hz, 640, 400 );
	//VGAController.setResolution(VGA_640x240_60Hz);    // select to have more free memory

	// this speed-up display but may generate flickering
	VGAController.enableBackgroundPrimitiveExecution( false );

}

// the loop function runs over and over again until power down or reset
void loop() 
{
	demo4();
}

void demo4()
{
	
	for( int i = 0; i < 500; ++i )
	{
		Canvas.setPenColor( random( 4 ), random( 4 ), random( 4 ) );
		Canvas.setPixel( random( 640 ), random( 350 ) );
		delay( 15 );
	}
	delay( 500 );
	
	for( int i = 0; i < 50; ++i )
	{
		Canvas.setPenColor( random( 4 ), random( 4 ), random( 4 ) );
		Canvas.drawLine( random( 640 ), random( 350 ), random( 640 ), random( 350 ) );
		delay( 50 );
	}
	delay( 500 );
	
	for( int i = 0; i < 50; ++i )
	{
		Canvas.setPenColor( random( 4 ), random( 4 ), random( 4 ) );
		Canvas.drawRectangle( random( 640 ), random( 350 ), random( 640 ), random( 350 ) );
		delay( 50 );
	}
	delay( 500 );
	
	for( int i = 0; i < 50; ++i )
	{
		Canvas.setPenColor( random( 4 ), random( 4 ), random( 4 ) );
		Canvas.drawEllipse( random( 640 ), random( 350 ), random( 640 ), random( 350 ) );
		delay( 50 );
	}

}
```

bytefresser, 2019-05-30:
Select 8 or 64color configuration. Check the first lines in the example.

YordanYanakiev, 2019-05-31:
I've did it as noted, yet I can not make it works.
it is working flawlessly on other similar library with the same
configuration anyway.

fdivitto, 2019-05-31:
>it is working flawlessly on other similar library with the same configuration anyway.
> […](#)

...for example? Which library? 
Can you post the schema you have used?

YordanYanakiev, 2019-06-01:
Sorry for the late answer fdvitto.
Here the circuit i am using:
[image]

As well I have a question: Can we use the library in a thread working on 1st or second core with xTaskCreatePinnedToCore ?

fdivitto, 2019-06-04:
You've chosen 8 colors mode, so you have to set:

#define USE_8_COLORS  1
#define USE_64_COLORS 0

YordanYanakiev, 2019-06-04:
It was set on 8 color mode. just when i were copy-pasting probably didn't
copy it.

fdivitto, 2019-06-04:
Ok, at this point I just don't know why it doesn't work and you haven't said yet which "similar" library works.
There are several possibilities still to explore: one of these is to check the ESP32 revision number. It must be rev-1 (not 0).

YordanYanakiev, 2019-06-05:
I am using  "https://github.com/bitluni/ESP32VGA" with the very same config for  bit color.
The revision of the SDK is 2

fdivitto, 2019-06-05:
I meant chip revision (rev0 doesn't work, rev 1 should work).

YordanYanakiev, 2019-06-05:
chip_info.revision returns 1

fdivitto, 2019-06-05:
In my board I have a led connected to PIN2 (where you have VSYNC): this doesn't work for me, maybe because the LED circuit modifies VSYNC signal. This is the reason I suggest to avoid to use that PIN in (look at the bottom of the page):

[http://www.fabglib.org/conf_v_g_a.html](http://www.fabglib.org/conf_v_g_a.html)

Could this be the reason?

YordanYanakiev, 2019-06-05:
well.. it works with the other lib.. hmm need to try somehow..
the led is blinking as a mad while it is working with the other lib, but it works.. hmm.

closed 2019-07-16 by fdivitto