fdivitto/FabGL #15: Example/UI mouse cannot be controlled

LilyGO, 2019-05-31:
[image]
I tried Mouse to work independently and the screen was full of images. But I can't control the cursor.
Of course, I have already confirmed in another Mouse_demo that the mouse can be used normally. look forward to your reply.
Finally, this is an elegant and interesting project. Thank you

fdivitto, 2019-05-31:
Which example are you testing? Please can you download current version and try the UI demo?

I'm sure you have already done this, but I have to tell: have you put the mouse on port 0 (where is the keyboard)?

LilyGO, 2019-06-04:
I found the place where the problem is.Change port 0 to port 1,The mouse works fine

But I still can't get the keyboard and mouse to work at the same time.
Do you have a program recommendation?

fdivitto, 2019-06-04:
To have both mouse and keyboard set:
```
#define KEYBOARD_ON_PORT0                0
#define MOUSE_ON_PORT0                   0
#define KEYBOARD_ON_PORT0_MOUSE_ON_PORT1 1
```

connect keyboard on port 0 and mouse on port 1. A demo that uses both is UI (still in developing state) or the Space Invaders.

LilyGO, 2019-06-06:
The first time I chose this option was in Space Invaders., but my mouse didn't work.

fdivitto, 2019-06-06:
please can you send me the schema of your board (privately if you prefer)?

LilyGO, 2019-06-06:
OK, I will send you the schematic.

closed 2019-06-06 by LilyGO