fdivitto/FabGL #16: uiApp in combination with wifi

zarya, 2019-07-13:
When wifi is enabled the code crashes on a assert

To reproduce this just add WiFi.begin(ssid,password); to the GraphicalUserInterface example

```
0x4008b43b: xQueueGenericSend at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/queue.c:1996
0x4008f060: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c:707
0x4008f291: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c:707
0x4008b43b: xQueueGenericSend at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/queue.c:1996
0x400d2eb6: fabgl::uiApp::postEvent(fabgl::uiEvent const*) at ??:?
0x400d3013: fabgl::uiApp::showWindow(fabgl::uiWindow*, bool) at ??:?
0x400d56bf: fabgl::uiApp::run() at ??:?
0x400d2451: loop() at ??:?
0x400db119: loopTask(void*) at ??:?
0x4008abb9: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c:355 (discriminator 1)
```

RegyRegy, 2019-07-19:
Have the same problem:
E (60) wifi: create wifi task: failed to create task
E (61) wifi: wifi_init_in_caller_task 713 ret=257

Network terminal does not work :(

RegyRegy, 2019-07-19:
The same problem is with BT:

rst:0xc (SW_CPU_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:8424
ho 0 tail 12 room 4
load:0x40080400,len:5868
entry 0x4008069c
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/queue.c:719 (xQueueGenericSend)- assert failed!
abort() was called at PC 0x40090da7 on core 1

Backtrace: 0x40094904:0x3ffcde20 0x40094b35:0x3ffcde40 0x40090da7:0x3ffcde60 0x400d2e56:0x3ffcdea0 0x401b03c1:0x3ffcded0 0x400d76fa:0x3ffcdef0 0x400d2fc1:0x3ffcdf90 0x400d1cd3:0x3ffcdfb0 0x400d83ff:0x3ffcdff0 0x40090525:0x3ffce010

Rebooting...

fdivitto, 2019-07-19:
I don't think there is enough free memory to be able to use WiFi. Try to reduce resolution to 640x200:
VGAController.setResolution(VGA_640x200_70Hz);

RegyRegy, 2019-07-19:
Yes, thats the point.
Reducing resolution allows WiFi to work,
other way there is no memory for that.

zarya, 2019-07-20:
Confirmed 640x350 is to much in combination with wifi
640x200_70Hz works indeed fine.

closed 2019-07-20 by zarya