fdivitto/FabGL #19: Speed up Pixel Writes

paulscottrobson, 2019-07-16:
Hi. I was looking at the PIXEL/PIXELINROW Macros considering the possibility of making them visible for faster access.  It dawned on me that all your remapping is doing, I think (?) is simply toggling bit 1 ; hence this could be done simply by ^2 (see below). I haven't tried it exhaustively but it works on what I'm doing (VGA 320x200), and I would doubt the optimiser would work out the equivalence. If it works it would probably be worth a few CPU cycles as it's used quite a lot AFAICS.

(this is from vgacontroller.c obviously)
```
// pixel 0 = byte 2, pixel 1 = byte 3, pixel 2 = byte 0, pixel 3 = byte 1 :
// pixel : 0  1  2  3  4  5  6  7  8  9 10 11 ...etc...
// byte  : 2  3  0  1  6  7  4  5 10 11  8  9 ...etc...
// dword : 0           1           2          ...etc...

//#define PIXELINROW(row, X) (row[((X) & 0xFFFC) + ((2 + (X)) & 3)])
#define PIXELINROW(row, X) (row[(X) ^ 2])
```

fdivitto, 2019-07-16:
wow, this is the kind of comments I love!! It just works, and the performance improvement is actually visible (look at double buffer demo).
Can I add your name in the vgacontroller source?

paulscottrobson, 2019-07-16:
No problem - as long as you still get 99% of the credit for this awesome piece of work :)

closed 2019-07-19 by fdivitto