fdivitto/FabGL #21: Suggestions?

rog77, 2019-07-27:
Sorry if you've heard this a million times already, but...

Is there any way you might consider implementing an xserver and ssh client? Also, I appreciate that this is a VGA library, but other ESP hacks have implemented video out over Composite Video, and analogue broadcast - is it possible to add this? These options would allow your project to bring ultra low cost computing to virtually anyone in the world, on variety of available displays - end users could get free BSD accounts on SDF or the like.

Either way, thanks for considering it!

fdivitto, 2019-08-04:
Composite video (black/white) may be added in future. Other (ssh, xserver,...) is out of topic for this library,

closed 2019-08-04 by fdivitto

stroobandt, 2019-10-12:
@fdivitto I agree with @rog77 An `ssh` secure command line is what would open this project to the world.

stroobandt, 2019-10-12:
@fdivitto Here is the [source code](http://www.xk72.com/midpssh/v1.7.3/midpssh.zip) of [MidpSSH](http://www.xk72.com/midpssh/), a mobile Java2ME SSH client which runs/ran on old S40 Nokia dumbphones. It can still be used to securely login on modern GNU/Linux machines. This might provide some inspiration...