fdivitto/FabGL #23: VGA looses sync after a few seconds..

WacKEDmaN, 2019-08-04:
using latest code from here.. 
upload NetworkTerminal.ino to esp32.. code loads fine and displays a screen...moments later screen starts to shimmer,then gains horizontal lines and then loses sync... reboot it gives same results...

i also think i  just killed my ps/2 keyboard with this latest code....it reported as OK... but wouldnt respond..no LEDs lit...pulled the keyboard apart and the IC is RED HOT... (ive triple checked the pinouts and they are perfect!).. 

so im sad now that i can no longer use or recommend this library :(

fdivitto, 2019-08-04:
I'm sad to hear this and of course I cannot replicate these problems. Anyway I have some questions:
- which board are you using? In case you built yourself, can I see the schema?
- which Arduino ESP32 version library are you using? (1.0.2, 1.0.2-rc2, 1.0.3-rc1...etc...)
- VGA loses sync after connecting to WiFi, or just few seconds after boot?

WacKEDmaN, 2019-08-04:
thanks for the quick reply!..
i think i found the issue...looks like two of my breadboard voltage regulators died..its outputting 10v on the 5v rail (with 12v input)... it was working perfectly fine last week!!..i ran the 8800 on it for a few days...
it didnt take out the ESP32...but it did take out the keyboard...constant error now... 
VGA has cleared up aswell i think it might have been getting a bit too much voltage back on the ps/2 pins!

FYI im using:
Doit ESP32 devkit v1 board
ESP32 lib v 1.0.2

but now no ps/2 keyboard...(mouse works!)
i guess ill have to setup so way of serial input

closed 2019-08-04 by WacKEDmaN