fdivitto/FabGL #24: Further VGA Board

roboticboyer, 2019-08-07:
There is a further ESP32 VGA Board: [https://www.tindie.com/products/bitluni/esp32-vga-black-edition/](https://www.tindie.com/products/bitluni/esp32-vga-black-edition/)

fdivitto, 2019-08-10:
Have you tested this board with FabGL? Can you confirm it works with FabGL?

roboticboyer, 2019-08-11:
Sorry, not yet
Please note that the Bitluni board is not equipped with mouse and keyboard ports, only the VGA adapter and resistors

mcuadros, 2019-08-13:
I made my own BTW. https://easyeda.com/mcuadros/fabgl-lolind32 and printed in JLCPCB for the price of 24,36e  10uds.

Works perfects besides I forgot link a couple of GND pins of the VGA. I also have a 3d printed box. 
[image]

kishima, 2019-08-13:
I'm also making my FabGL board for ESP32-WROVER-B and a binding for mruby.

[image]

https://github.com/kishima/mruby-esp32-narya

For building mruby, I made a FabGL component for ESP-IDF. It can be build with Arduino core for ESP32.

https://github.com/kishima/FabGL_component

linker3000, 2019-08-20:
Has anyone added sound to Space Invaders? I'd like to demo the ESP32 at a local STEM event and that would be an attention-getter, but I'm not smart enough to modify and test the code in the time before the event!

linker3000, 2019-08-20:
Breadboarders might be interested in this VGA+keyboard breakout board I laid out.

https://easyeda.com/linker3000/vga-ps2-breakout

WacKEDmaN, 2019-08-22:
nice one @linker3000 ...needs dual PS/2 tho for keyboard and mouse....
im using a dual stacked header from an old motherboard

linker3000, 2019-08-23:
@WacKEDmaN Yep. My board was laid out for some retro Z80 work hence no mouse port.

douggilliland, 2020-01-26:
I made one, too.

http://land-boards.com/blwiki/index.php?title=ESP32-VGA

[image]

fdivitto, 2020-02-07:
This is my developing board with micro SD card and all pins exposed (also on [tindie](https://www.tindie.com/products/fabgl/fabgl-development-board/)):

[image]

[image]

kishima, 2020-02-08:
I have updated my board. It supports UART with a GROVE connector and SD card as well.

[image]

I also made an editor for Ruby with fabgl::Terminal class. Now I can write and run Ruby code on the board.
[image]

[My project site](https://kishima.github.io/family_mruby/)
I appreciate this awesome library.

AlyssonRowan, 2020-06-12:
An excellent library which hits all of the high points and landmarks for a video terminal.

I am basing a micro terminal project on this library, and on the circuit designs. I only need text mode support, my target being a terminal for my ancient PDP-11 and for an 8052 baed system.

[My projects Blog](https://alyssonrowanprojects.blogspot.com/2020/06/vga-text-terminals-almost-off-shelf.html)

PQCraft, 2020-10-13:
If any of you wants to test your boards might, I am working on a version of BASIC specifically for the ESP32 and FabGL: [https://github.com/PQCraft/ESPBASIC](https://github.com/PQCraft/ESPBASIC). Keep in mind that it is still in beta. It has awkward bugs that sometimes requires me to just rewrite the entire routine. I also need people to tell me what the bugs are.

AlyssonRowan, 2020-10-13:
Now that is very interesting. I am still working on my terminal, so I won't be in a position to try this at the moment.

I can see this idea ending up as a replacement for my 8052 microcontroller boards.

douggilliland, 2021-01-29:
I made a [new version of the card with the SD card socket.](land-boards.com/blwiki/index.php?title=ESP32-VGA) 
[image]

fdivitto, 2021-08-10:
### [FabGL Development Board with I/O Extender and PSRAM](https://www.tindie.com/products/24612/)
#### Features:
- **WROVER-E** based with **16MB Flash** and **4MB PSRAM**
- **VGA** video output
- **Audio** output (line level)
- **PS/2 Mouse** and **Keyboard** inputs
- **Micro SD** socket
- I/O extender with **16 GPIOs** (3.3V max voltage, MCP23S17 based)
- firmware upload up to 460800 bps (uses CH340C USB-Serial chip)

Supports all FabGL applications, including the PC Emulator: now you can program your I/O pins using GWBasic or QBASIC under MSDOS!!

Compatible with [LEDs and Buttons board](https://www.tindie.com/products/24613/)

[Purchasing this board](https://www.tindie.com/products/24612/) you support the FabGL project!

Examples with pluggable LEDs and Buttons board:
- [Video Example 1](https://www.youtube.com/watch?v=O9AP4vZHfjI)
- [Video Example 2](https://www.youtube.com/watch?v=JrENR8zDtyU)

[image]

fdivitto, 2021-10-04:
from now on please post new boards into "discussions"

closed 2021-10-04 by fdivitto