fdivitto/FabGL #26: How to use the USB port ?

markbirss, 2019-08-29:
Next to the microUSB port is a USB port, how to use it ?

fdivitto, 2019-09-02:
Please specify the board you are using. This project is not related to any specific board.

markbirss, 2019-09-02:
Ok, understood

Just for reference im using the TTGO VGA32 V1.1 which has been upgraded to V1.2 to include JST and mini stereo connector

[image]

https://www.aliexpress.com/item/4000121438408.html
https://www.aliexpress.com/i/33039492191.html

fdivitto, 2019-09-02:
There is only one USB port. Next to the micro USB there are exposed GPIO12 and GPIO2 that you can use as second UART (for example to use the board as ANSI/VT terminal for a Linux/Unix system).

markbirss, 2019-09-02:
Ok,

Thank you, I see

I wanted to run that example but was not sure where to connec the RX and TX.

I used the Network Terminal Telnet instead but my 80x25 screen does not fit
correctly

Is this this issue with the example or possibly with the image correction
on my screen (VGA adjust), you would need 640x480 or 800x600 for a full
80x25 screen to fit ?

fdivitto, 2019-09-03:
Have you tried to adjust screen position with:

VGAController.moveScreen(20, ...put here a value, for example -5 or +5...);

markbirss, 2019-09-03:
No not yet, but will look into that.

Thank u

closed 2019-12-18 by fdivitto