fdivitto/FabGL #33: Missing fonts on ESP32

linker3000, 2019-12-04:
I quickly compiled the 0.8.0 build of invaders last night for an ESP32 dev board. The only code change I make for this is to add a line in the setup code section to identify the right VGA output pins.

The game runs as expected, but on the opening page the text giving the points value for all the game objects doesn't appear.

fdivitto, 2019-12-04:
Are you using 64 colors or 8 colors configuration? Can I see the changes you made in the setup section?
Meantime can you try to change line 106 from:
   canvas.setPenColor(Color::White);
to
  canvas.setPenColor(Color::BrightWhite);

linker3000, 2019-12-05:
Hi Fabio,
Thanks for the quick reply. I am only using 8 colors right now.
The config line I need for my setup, and a change of video mode (I have tried both) is:

```
void setup()
{
  PS2Controller.begin(PS2Preset::KeyboardPort0_MousePort1, KbdMode::GenerateVirtualKeys);

  //VGAController.begin();
  VGAController.begin(GPIO_NUM_22, GPIO_NUM_21, GPIO_NUM_19, GPIO_NUM_18, GPIO_NUM_5);
  VGAController.setResolution(QVGA_320x240_60Hz);

  //VGAController.setResolution(VGA_320x200_75Hz);

[snip]

```

Yep, setting BrightWhite fixes the issue.

closed 2019-12-05 by linker3000