fdivitto/FabGL #35: TTGO schematic?

douggilliland, 2019-12-06:
I bought two of the TTGO builds of this. Very nice. Works well. Trying to figure out the GPIO pins. Most are on the silkscreen. One switch isn't listed. Is there a schematic out there for the TTGO build?

fdivitto, 2019-12-06:
Please ask TTGO for the schematics.

douggilliland, 2019-12-06:
Your project is fantastic. Not sure why more people don't understand it. A $12USD ANSI terminal is the killer design for Retro-Computer fans. Keep up the good work.

fdivitto, 2019-12-06:
Thank you! Actually my first prototype did cost even less. For about 5$ you can purchase an ESP32 board with USB. Add just few resistors looking at this "generic" [schema](http://www.fabglib.org/gen_schema.html).

douggilliland, 2019-12-06:
I did see the schematic on your site. (As you know) the TTGO design went with different I/O connections. I do think the TTGO card is a good value for money considering it has audio, VGA and 2xPS/2 connectors. 

The only thing about the TTGO card I don't like is the lack of mounting holes.

fdivitto, 2019-12-06:
There is no difference in terms of GPIOs used: TTGO uses the same pin assignments of my schema. TTGO uses better level shifters (while I use resistors!!) for PS/2 ports. Their board also has external battery support.

douggilliland, 2019-12-06:
I am trying to do the ANSI terminal. I want to connect it to the 4 pin header on the TTGO card.

Their 4 pin header connects to IO12 and IO2.

Your schematic use IO2 for TX and IO14 for RX.

douggilliland, 2019-12-06:
[image]

fdivitto, 2019-12-07:
You're right, I've just fixed that schema! The unique place where FabGL uses serial RX/TX is in AnsiTerminal example and it uses GPIO12 as RX and GPIO2 as TX.

Thank you!

closed 2019-12-18 by fdivitto