fdivitto/FabGL #37: NetworkTerminal wifi wont connect...

WacKEDmaN, 2019-12-27:
im having issues connecting to Wifi SSID with the NetworkTerminal.ino 
scan shows the SSID...
type wifi ssid pass1234
tries to connect but fails...
it seems the ssid and psw strings are receiving the correct details...but still no connect...

BUT...if i add the WiFi.begin(SSID,pass1234); to setup() then it connects every time...
any thoughts?

fdivitto, 2019-12-31:
Please try to change exe_wifi() code as follows:
```
void exe_wifi()
{
  static const int MAX_SSID_SIZE = 32;
  static const int MAX_PSW_SIZE  = 32;
  char ssid[MAX_SSID_SIZE + 1];
  char psw[MAX_PSW_SIZE + 1] = {0};
  error = true;
  if (sscanf(inputLine, "wifi %32s %32s", ssid, psw) >= 1) {
    Terminal.write("Connecting WiFi...");
    Terminal.flush();
    fabgl::suspendInterrupts();
    WiFi.disconnect(true, true);
    for (int i = 0; i < 2; ++i) {
      WiFi.begin(ssid, psw);
      if (WiFi.waitForConnectResult() == WL_CONNECTED)
        break;
      WiFi.disconnect(true, true);
    }
    fabgl::resumeInterrupts();
    if (WiFi.status() == WL_CONNECTED) {
      Terminal.printf("connected to %s, IP is %s\r\n", WiFi.SSID().c_str(), WiFi.localIP().toString().c_str());
      error = false;
    } else {
      Terminal.write("failed!\r\n");
    }
  }
  state = State::Prompt;
}
```

WacKEDmaN, 2019-12-31:
thanks Fab...
ill have to get back to you...seems i killed my esp32!... enabling wifi now (with any sketch) causes it to reboot... they are not as robust as the esp8266 chips! (ive killed 3 esp32s now!)

fdivitto, 2019-12-31:
It is strange, I did everything and sometimes I even reversed the polarity and shorted the outputs, but I never broke one!

Have you tried to full erase ESP32 flash?
The general command should be something like:

python esptool.py erase_flash

but it may vary depending by your system.

WacKEDmaN, 2019-12-31:
yup tried erase on all 3 (2x esp32-cam 1x devkit v1) 
one of the esp32-cams has issues with the wifi being very weak (i need to put my tablet right on top of the antenna to get a connection...move it away 1cm and the connection drops! (everything else works fine)... the other esp32-cam has major issues cant get a serial connection to it at all...
the esp32 devkit (which i was using with your library) now drops serial connection and reboots if i enable wifi (havent tried with bluetooth)... sketches that dont use wifi seem to work as they should (eg blink)
im thinking the devkit board was actually damaged when my power supply over volted (15v instead of 5v!) and killed my keyboard a few months back!

thanks again.... ill let you know how the new board goes when i get it...im trying to find a dev board with 4 or 8mb PSRAM onboard (eg ESP-WROVER)...then we'll have RAM for Amstrad CPC emu! :P

closed 2019-12-31 by WacKEDmaN

WacKEDmaN, 2020-01-20:
got a hold of a TTGO-Energy...
wifi working now... even tho it says authentication failed!... few seconds later it resets the wifi and connects!
heres the verbose log...
```
[D][esp32-hal-psram.c:47] psramInit(): PSRAM enabled
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 1 - SCAN_DONE
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 202 - AUTH_FAIL
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 205 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 192.168.1.36, MASK: 255.255.255.0, GW: 192.168.1.1

```
i think the error on the original board was just due to power... when i hooked up the TTGO-E with 5v to PS/2 ports via the board i was getting the same core panic'ed error as above.... when i changed it to the onboard 3.3v rail... it worked fine...(including the keyboard and mouse)

now the fun begins trying to load roms into the PSRAM! :P