fdivitto/FabGL #38: C64 Emulator?

knopserl, 2020-01-08:


knopserl, 2020-01-08:
great project. I ordered two TTGO VGA32. I have also seen the VIC20 emulator. Amzing.
However it would be very very nice to have also a C64 emulator (that would be amazing).
I have seen a github project with an C64 emulator on an ESP8266 but with a Web-UI. 
https://github.com/NardJ/esp8266-C64/issues/6 
So would there be a possibility to combine that? Actually there was a request in the other direction (ESP8266 C64 emulator should use the FabGL lib) but I think having it on ESP32 would be better.

fdivitto, 2020-01-09:
The linked emulator (https://github.com/NardJ/esp8266-C64) isn't actually a full working emulator. It just executes 6502 code and checks what the code writes on the C64 screen buffer, sending that to a browser. The "code" of course is the C64 kernal and basic ROMs. What an emulator should do is to emulate the video chip, the audio chip, the timer chips and synchronize them at CPU cycle level. It is necessary when you run games that use 100% of video, audio, timers and CPU chips.
It is very complex to emulate VICII (the C64 video chip) and SID when you have so low resources as you have on ESP32 or ESP8266. Of course nothing is impossible, but it requires a lot of work!
At the other side if you just need to run very simple Basic programs like in the linked project it is a trivial task and in one day of work it should be possible, but someone may ask you why so few games run on your emulator!

closed 2020-01-16 by fdivitto