fdivitto/FabGL #40: Space invaders

CornN64, 2020-01-10:
Hi,
Impressive work on this library!
I noticed a small bug/issue on space invaders that you might consider fixing.
The enemies should bounce (eg. turn direction) only when they one closest to the screen edge actually reaches the edge of the screen. Right now it seems to remember the position of already destroyed enemies and bounce way to early.
Keep up the nice work.
Cheers

douggilliland, 2020-01-11:
This isn't a bug, but I ran the ESP32 Space Invaders on my own hardware and it worked great. https://www.youtube.com/watch?v=TK8Gp9_7ZdE

CornN64, 2020-01-11:
I think you misunderstood what I meant (yes the game runs but the game mechanics are a bit off)
The direction of the aliens should only change direction once they hit the screen border but you will notice that if you completely remove aliens in the leftmost columns they will still change direction way to early and before hitting the screen edge (as if the removed aliens are still alive).
I hope this description makes more sense.
Cheers

fdivitto, 2020-01-16:
just fixed!
I've also added sounds to Space Invaders.

closed 2020-01-16 by fdivitto

WacKEDmaN, 2020-01-20:
nice Fab... i was looking into adding sounds to space invaders before my old board died!