fdivitto/FabGL #43: Space Invaders Build error

douggilliland, 2020-01-29:
I tried to build space invaders and got a build error.

```
> C:\Users\HPz420\Documents\GitHub\Other People GitHubs\FabGL VGA on ESP32\FabGL\examples\VGA\SpaceInvaders\SpaceInvaders.ino: In member function 'virtual void IntroScene::init()':
> 
> SpaceInvaders:90:29: error: 'class fabgl::SoundGenerator' has no member named 'playSamples'
> 
>      music_ = soundGenerator.playSamples(themeSoundSamples, sizeof(themeSoundSamples), 100, -1);
```
Might be due to installing FabGL via the Arduino IDE and it's using the library it loaded from there but not sure how to solve this.

fdivitto, 2020-01-29:
Make sure you're using the library is currently on github instead of the one from library manager.

douggilliland, 2020-01-30:
I don't know how to do that. Do I uninstall the FabGL that I installed from library manager? How do I then point to this one?

douggilliland, 2020-01-31:
I figured it out. If anyone else is stuck on this...
1 - Install drivers via Arduino from the Library Manager. These are not the latest files. Space Invaders should run but not have sound (as of now because it's the older version).
2 - Move the FabGL from the Arduino libraries file folder to the desktop. 
3 - Copy the FabGL from the GitHub download to the Arduino files folder (where you removed the other folder).
4 - Examples will now be the latest.
5 - Compile/download/test the new files. You know it works if you now get sound from Space Invaders.
6- Save as where you would normally save your Arduino sketches.

closed 2020-02-18 by douggilliland