fdivitto/FabGL #45: Space Invaders freeze

rdkluijv, 2020-02-05:
Hi,

Thanks for the great work.
Space Invaders sometimes freezes in the latest version. Probable cause:
The if statement for the touch sides should be true two times to change direction. This is not necessarily the case. The lastHitEnemy could make the sprite causing the touch invisible. So no touch the second time because the move checks the visibility.
Possible solution: A small finite state machine that prepares the next move. I could write it for you if you like.

Regards,

Bor.

fdivitto, 2020-02-06:
Hi,
thank you for the feedback. I've experienced some crashes (probably caused by the audio engine, I still have to investigate...), but not "freezes". Have you a screen shot or a video of the problem?

rdkluijv, 2020-02-06:
No screenshot and hard to reproduce.
Description: 
When firing at the utmost left of the screen the enemy sprites only go down and don't start moving to the right. The rest of the game functions as before.
Problem is here:
        if (touchLeftSide || touchRightSide) {
          if (enemiesDir_ == 0) {
            enemiesDir_ = touchLeftSide ? 1 : -1;
          } else {
            enemiesDir_ = 0;
            enemiesY_ += ENEMIES_STEP_Y;
          }
        }
This needs 2 updates to work but touchLeftSide can be false the second time because of:
        if (lastHitEnemy_) {
          lastHitEnemy_->visible = false;
          lastHitEnemy_ = nullptr;
        }
The result being that enemiesDir_ stays zero. So no more movement.
Something like this should fix it:
        // Calculate next point.
        if (touched && step_y) {
            step_y = false;
            enemiesY_ += ENEMIES_STEP_Y;
            moveIncrement = !moveIncrement;
        } else {
          step_y = true;
          if (moveIncrement) {
            enemiesX_ += ENEMIES_STEP_X;
          } else {
            enemiesX_ -= ENEMIES_STEP_X;
          }
        }
I made an example but I didn't build it. It's also 7z instead of zip.
[SpaceInvaders.zip](https://github.com/fdivitto/FabGL/files/4164606/SpaceInvaders.zip)
Hope this explains it.

fdivitto, 2020-02-07:
thank you, now understand. My last commit should fix the problem.

closed 2020-02-07 by fdivitto