fdivitto/FabGL #46: Request: Mouse only port 1 instead of 0

rdkluijv, 2020-02-05:
Hi,

In order to match the colors of the connectors it makes more sense to use port 1 if only the mouse is connected. Just a change in the enum and the GPIO matrix switch for the pin numbers. Again I could make the changes.

Regards,

Bor.

fdivitto, 2021-03-13:
Sorry for the big delay! I don't understand this question, port 0 is the primary input (keyboard) - lavender color, port 1 is the optional secondary input (mouse), which is green.

rdkluijv, 2021-03-13:
No problem. I used a workaround.
The small issue is with the PS2Preset enum. If you have no keyboard just a mouse it's mapped to port 0. That's fine and it works but it might be better to map it to port 1. Reason: Because of the color of the connector port 1 is supposed the mouse connector.
These are the options not in the enum:
KeyboardPort1_MousePort0
KeyboardPort1
MousePort1
Only the last would make sense to me. It is also possible to have 2 keyboards or 2 mouses. Then the port does not matter.

fdivitto, 2021-03-14:
OK, now understand. Now can be "feature request"...

rdkluijv, 2021-03-14:
Thanks. It's just to make programs more readable. BTW I created a pong version with the software and there I needed 2 mouses.

fdivitto, 2021-03-14:
Thank you, just added!

closed 2021-03-14 by fdivitto