fdivitto/FabGL #48: VGA schematic issue

douggilliland, 2020-02-17:
Your reference schematic uses 400/800 ohm resistors. That doesn't provide full scale VGA. I am using 270/560 ohms to get the full 1V VGA signal. The ESP32 can drive the 13 mA with a lot of room to spare. Here's the [LT Spice simulation I made](http://land-boards.com/blwiki/index.php?title=ESP32-VGA#VGA_-_Ideal_Drive_2:2:2_Case).

Also your table shows the pins reversed from the schematic resistor values. For instance, R0 is connected to GP22 and R1 is connected to R1. If R1 is the more significant bit it should go to the lower value resistor but it's shown going to the 800 ohm resistor in the schematic.

[image]

douggilliland, 2020-02-17:
The docs could have it backwards in some places. For instance: http://www.fabglib.org/classfabgl_1_1_v_g_a_controller.html#a11b70942a5879a6241dea608b28a65b8

douggilliland, 2020-02-17:
It contradicts this page:

https://github.com/fdivitto/FabGL/blob/master/fabgl%20GPIOs%20assignment.txt

VGA
  HSync => 23
  VSync => 15
  R0    => 22
  R1    => 21
  G0    => 19
  G1    => 18
  B0    => 5
  B1    => 4

bootrino, 2020-02-17:
What do you mean by "full scale VGA"?

douggilliland, 2020-02-17:
VGA analog signals go from 0V to 1V (full-scale). There is a 75 ohm terminator in the VGA device. 3.3V out and the resistor values make a voltage divider with the load resistor in the VGA device.

douggilliland, 2020-02-17:
Screen brightness is the difference. I switched to the values above and get a much brighter screen. Still within the VGA specification.

douggilliland, 2020-02-17:
Reading more. The proper peak voltage should be 0.7V for video. I will increase the resistor values to match but 400 is way too high for brightest video.

http://www.microvga.com/faq/electrical/what-are-vga-voltage-levels

douggilliland, 2020-02-17:
I wonder if it's my monitor? The above page says some like 1V peak, others 0.7V peak. Mine tolerates the 1V well. Maybe the 400/800 works best for 400/800.

douggilliland, 2020-02-17:
I think just this page might be the backwards part:

https://github.com/fdivitto/FabGL/blob/master/fabgl%20GPIOs%20assignment.txt

fdivitto, 2020-12-09:
fixed thanks

closed 2020-12-09 by fdivitto