fdivitto/FabGL #49: SD Card not working with FabGL?

Kaef, 2020-03-03:
Hi,
thank you very much sharing this impressive library.
I try to use the esp32 sd-card in combination with FabGL, but the sd-card does not work.
Does FabGL uses any resources needed by the sd-card subsystem?
If so, would it be possible to solve this issue?
Best regards,
Kaef

fdivitto, 2020-03-03:
Hi,
please add details. Which board are you using, which pins are connected to the SD Card?

Kaef, 2020-03-03:
I'm using a TTGO-T8 board, sdcard pins are:
    CS - IO13
    MOSI - IO15
    SCK - IO 14
    MISO - IO2
sdcard is working if I don't use FabGL. When using FabGL the card can not be mounted.

fdivitto, 2020-03-03:
IO15 is used for VSync

Kaef, 2020-03-03:
Sorry, missing info.
I use:
    #include <fabgl.h>
    fabgl::VGAController VGAController;
    fabgl::PS2Controller PS2Controller;
    fabgl::Terminal      Terminal;

and in setup() I call:
    VGAController.begin(GPIO_NUM_25, GPIO_NUM_26, GPIO_NUM_27, GPIO_NUM_5, GPIO_NUM_23);

fdivitto, 2020-03-03:
is this the board are you using?
https://www.aliexpress.com/item/32911178069.html

If so, I should have one to test.

Kaef, 2020-03-03:
yes, nearly; on my board  "T8 V 1.7.1, 20190702" is printed

Kaef, 2020-03-03:
Do you have tested another esp32 board using FabGL and sdcard?

fdivitto, 2020-03-03:
Just the prototypes of my board:
https://www.tindie.com/products/fabgl/fabgl-development-board/

Kaef, 2020-03-03:
Which pins do you use for sdcard?

fdivitto, 2020-03-03:
The fabgl defaults. They are listed here:

https://github.com/fdivitto/FabGL/blob/master/fabgl%20GPIOs%20assignment.txt

Kaef, 2020-03-03:
Thank you very much, I try using the pins you have used and will report if it works.

Kaef, 2020-03-11:
I can confirm that using the fabgl defaults the sdcard is working.
When using my gpio definitions my test program is also working fine. So I think the problem is in my code, sorry.

Kaef, 2020-03-12:
Problem solved!
Mounting the sdcard failed because free memory was too low, reducing the FabGl resolutions solves the problem.

closed 2020-03-12 by Kaef