fdivitto/FabGL #51: Fabgl PS2Controller.begin and Rebooting

elbow, 2020-03-08:
I have a devkit v1 with bitluni's 14 bit VGA wiring.

FabGL VGA out works fine if I map to the most significant bits for each colour.

But PS2Controller just causes my ESP32 to crash and reboot over and over.

14 bit vga uses up a lot of pins, so there's not much left.

I wired the audio output on D25 which is normal for fabgl.

And I hooked up D5 and D26 using the series resistor and the pull-ups per your schematic for the PS2 keyboard.

I modified one of the example scripts and initialised with

```PS2Controller.begin(GPIO_NUM_26, GPIO_NUM_5, GPIO_NUM_39, GPIO_NUM_39);```

The result is this, over and over:

```18:08:17.668 -> E (54) RTC_MODULE: /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/rtc_module.c:142 (rtc_gpio_init):RTC_GPIO number error
18:08:17.701 -> assertion "0 && "invalid register base"" failed: file "/Users/steve/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/ulp/esp32/ulp.h", line 299, function: uint32_t SOC_REG_TO_ULP_PERIPH_SEL(uint32_t)
18:08:17.701 -> abort() was called at PC 0x400e1e1b on core 1
18:08:17.701 -> 
18:08:17.701 -> Backtrace: 0x4008e428:0x3ffb1e70 0x4008e655:0x3ffb1e90 0x400e1e1b:0x3ffb1eb0 0x400d8242:0x3ffb1ee0 0x400d8345:0x3ffb1f00 0x400d85c5:0x3ffb1f50 0x400d1292:0x3ffb1f80 0x400dae27:0x3ffb1fb0 0x4008a895:0x3ffb1fd0
18:08:17.735 -> 
18:08:17.735 -> Rebooting...
```

Can anyone assist me in getting on track?

As I mentioned, the vga-only examples do work.

fdivitto, 2020-07-07:
Hello,
sorry for the delay.
Because PS/2 ports are handled by the ULP processor, only few pins are actually usable. They are:
GPIO_NUM_2
GPIO_NUM_4
GPIO_NUM_12 (with some limitations)
GPIO_NUM_13
GPIO_NUM_14
GPIO_NUM_15
GPIO_NUM_25
GPIO_NUM_26
GPIO_NUM_27
GPIO_NUM_32
GPIO_NUM_33

closed 2020-07-07 by fdivitto

elbow, 2020-07-07:
Hi,

Thanks for coming back.  I built your design too and it works great.  So now I have bitluni's with the 14 bit VGA and yours with less colours but with the PS2 interfaces.

Thanks.