fdivitto/FabGL #53: SPIFFS ?

paulscottrobson, 2020-03-15:
Hi. I am using SPIFFS in a fabgl project and it was terrible, randomly crashing. By chance I came across your Altair code which uses SPIFFS and I noticed all your calls were surrounded by interrupt disabling code. Copying this seems to make it work perfectly.

I was wondering if this is the problem - interrupts in SPIFFS break writing - if it is worth highlighting this in the documentation.

Many thanks

fdivitto, 2020-10-11:
Hi, last update fixes SPIFFS and interrupts problem, so suspending interrupts is no more required (it has been removed from all examples and internal code).
Of course this needs to be well tested, so your feedbacks are welcome!
Thanks

fdivitto, 2020-10-11:
please reopen this issue if you find problems. Thanks!

closed 2020-10-11 by fdivitto