fdivitto/FabGL #54: Simple Audio Example

asmaklad, 2020-03-27:
Hi There, 
I am trying to make this code works which is a strut from the Audio.ino example without the UI parts.

`#include "fabgl.h"
#include "fabui.h"

#include "speech.h"
#include "loop.h"
#include "mario.h"

SoundGenerator       soundGenerator;

void setup()
{
  Serial.begin(115200);
  Serial.println("#1");
  SineWaveformGenerator sine;
  SquareWaveformGenerator square;
  NoiseWaveformGenerator noise;
  TriangleWaveformGenerator triangle;
  SawtoothWaveformGenerator sawtooth;
  SamplesGenerator loop = SamplesGenerator(loopSamples, sizeof(loopSamples));
  SamplesGenerator speech = SamplesGenerator(speechSamples, sizeof(speechSamples));
  SamplesGenerator mario = SamplesGenerator(marioSamples, sizeof(marioSamples));
  WaveformGenerator * curGen = &mario;
  ///////////////////    
  Serial.println("#3");
  curGen->setVolume(100);
  //curGen->setSampleRate(160000);
  Serial.println("#4");
  soundGenerator.play(true);
  Serial.println("#5");
  soundGenerator.setVolume(90);
  Serial.println("#6");
  soundGenerator.detach(curGen);
  Serial.println("#2");
  soundGenerator.attach(curGen);
  curGen->enable(true);
  curGen->setFrequency(2520);  
  Serial.println("#7");
}

void loop()
{
  Serial.println("#8");
  delay(1000);
}`

however I get ESP32 to freakout and generate this error:

`7:23:42.811 -> ets Jun  8 2016 00:22:57
07:23:42.811 -> 
07:23:42.811 -> rst:0xc (SW_CPU_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
07:23:42.811 -> configsip: 188777542, SPIWP:0xee
07:23:42.811 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
07:23:42.811 -> mode:DIO, clock div:1
07:23:42.811 -> load:0x3fff0018,len:4
07:23:42.811 -> load:0x3fff001c,len:1216
07:23:42.811 -> ho 0 tail 12 room 4
07:23:42.811 -> load:0x40078000,len:9720
07:23:42.811 -> ho 0 tail 12 room 4
07:23:42.811 -> load:0x40080400,len:6352
07:23:42.811 -> entry 0x400806b8
07:23:42.981 -> #1
07:23:42.981 -> #3
07:23:42.981 -> #4
07:23:42.981 -> #5
07:23:42.981 -> #6
07:23:43.014 -> #2
07:23:43.048 -> #7
07:23:43.048 -> #8
07:23:43.115 -> Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
07:23:43.115 -> Core 0 register dump:
07:23:43.115 -> PC      : 0x400d1639  PS      : 0x00060830  A0      : 0x8008839c  A1      : 0x3ffb8770  
07:23:43.115 -> A2      : 0x3ffc054c  A3      : 0x3ffbac74  A4      : 0x00000000  A5      : 0x0000005a  
07:23:43.115 -> A6      : 0x3ffb1f78  A7      : 0x00000000  A8      : 0x00000001  A9      : 0x3ffb8750  
07:23:43.115 -> A10     : 0x3ffb1f78  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x3ffb878c  
07:23:43.115 -> A14     : 0xffffffff  A15     : 0x00000190  SAR     : 0x0000001a  EXCCAUSE: 0x0000001c  
07:23:43.115 -> EXCVADDR: 0x0000000d  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  
07:23:43.115 -> 
07:23:43.115 -> Backtrace: 0x400d1639:0x3ffb8770 0x40088399:0x3ffb87b0
07:23:43.115 -> 
07:23:43.115 -> Rebooting...`

Any clue on how to make this simple audio example work? the documentation says very little about the sequence of setting up the WaveFormGenerator and SoundGenerator classes. also the limits of Frequence.

paulscottrobson, 2020-03-30:
Hi. My code at https://github.com/paulscottrobson/eris/blob/master/esp32/src/main.cpp works on the same board. The control code is about line 220 and the setup code at line 280.

asmaklad, 2020-04-10:
for the sake of knowledge, it works when the WaveFormGenerators are moved outside setup into becoming global variables. Makes sense since The SoundGenerator is acting beyond the setup function and any variables declared inside setup will disappear from the memory once it is finished, while the SoundGenerator lives beyond it.
/*
  Created by Fabrizio Di Vittorio (fdivitto2013@gmail.com) - www.fabgl.com
  Copyright (c) 2019 Fabrizio Di Vittorio.
  All rights reserved.

  This file is part of FabGL Library.

  FabGL is free software: you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation, either version 3 of the License, or
  (at your option) any later version.

  FabGL is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU General Public License for more details.

  You should have received a copy of the GNU General Public License
  along with FabGL.  If not, see <http://www.gnu.org/licenses/>.
 */

#include <fabgl.h>

#define AUDIOCLOCK   (5000000L)

#include "speech.h"
#include "loop.h"
#include "mario.h"

SoundGenerator       soundGenerator;
SquareWaveformGenerator square;
SineWaveformGenerator sine;
NoiseWaveformGenerator noise;
TriangleWaveformGenerator triangle;
SawtoothWaveformGenerator sawtooth;
WaveformGenerator * curGen = &square;

void setup()
{
  Serial.begin(115200);
  Serial.println("#1");
  SamplesGenerator loopx = SamplesGenerator(loopSamples, sizeof(loopSamples));
  SamplesGenerator speech = SamplesGenerator(speechSamples, sizeof(speechSamples));
  SamplesGenerator mario = SamplesGenerator(marioSamples, sizeof(marioSamples));

  //ASM/////////////////    
  curGen=&sine;
  Serial.println("#2");
  curGen->setVolume(100);
  Serial.println("#3");
  curGen->setSampleRate(160000);  
  Serial.println("#5");
  soundGenerator.setVolume(90);
  Serial.println("#6");
  soundGenerator.detach(curGen);
  
  Serial.println("#7");  
  soundGenerator.attach(curGen);
  Serial.println("#8");  
  curGen->enable(false);
  Serial.println("#9");
  soundGenerator.play(true);
  Serial.println("#4");
  curGen->enable(true);
  curGen->setFrequency( (AUDIOCLOCK/7) ); 
}

void loop()
{
  
  Serial.println("#10");
  delay(1000);
}

closed 2020-04-10 by asmaklad