fdivitto/FabGL #56: Working 320x256 modeline?

richms, 2020-04-07:
I have some old LCDs here that are destined for the trash, they are 1280x1024 resolution.

Thought that I would try 1/4 the resolution in each way on it, but I cant seem to get a modeline that works. I tried dividing everything on this:

"1280x1024_60.00" 108.0  1280 1328 1440 1688 1024 1025 1028 1066 +hsync +vsync 

by 4 and it doesnt work. Anyone got a line that works ok?

If I dont have it at the right resolution then the screen pops up a warning about resolution which doesnt go away till the buttons on the screen are pressed, which is not ideal.

PQCraft, 2020-10-05:
I tried using a 256x192 (compatible with 1024x768) res and it wouldn't work. Similar issue.

fdivitto, 2020-10-12:
256x192 requires an upscaler, because it is incompatible with modern monitors. 

About 1/4 of 1280x1024, that should be 320x256 you may try this:
"\"320x256@120\" 12.6 320 328 376 400 256 262 263 278 -HSync -VSync DoubleScan FrontPorchBegins"
...but it depends by your monitor....

closed 2020-10-12 by fdivitto