fdivitto/FabGL #58: micropython library

calin2k, 2020-05-09:
Hi,
Is there a port of this library for micropython?

niutech, 2021-02-22:
I have a similar question. I would like to edit and run Micropython scripts on FabGL Dev Board using VGA output and PS/2 keyboard, similar to BASIC interpreter. There is [Micropython firmware for ESP32](https://micropython.org/download/esp32/), but how to link it with FabGL?

fdivitto, 2021-04-22:
This is interesting, but I don't have the time to do it. Are there clear instructions how to compile micropython for esp32?

niutech, 2021-04-23:
There is a [Github repo with instructions](https://github.com/micropython/micropython/tree/master/ports/esp32#readme). One option would be to [make FabGL a Micropython module](https://docs.micropython.org/en/latest/develop/extendingmicropython.html). Another one would be to [build Micropython for FreeDOS](https://github.com/micropython/micropython/wiki/Building-micropython-for-FreeDOS) (there is even [prebuilt upython.exe](https://github.com/pohmelie/micropython-freedos/releases)) and use it in PCEmulator.

kamo0829, 2021-05-02:
I've tried integrating Micropython and FabGL. If you are interested, you may visit my web site http://www.hi-ho.ne.jp/jr2xzy/cat/13-1.html. It is Japanese web site however you can see it by using google translation.

closed 2021-07-01 by fdivitto