fdivitto/FabGL #61: Kernel Panic

KrzysztofJeszke, 2020-05-20:
`ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Core 1 register dump:
PC      : 0x400819ef  PS      : 0x00060034  A0      : 0x80081cd2  A1      : 0x3ffbe750  
A2      : 0x3ffc15b8  A3      : 0x3ffbe780  A4      : 0x3ffba264  A5      : 0x0000abab  
A6      : 0x3ffc3a5c  A7      : 0x3ffba264  A8      : 0xbad00bad  A9      : 0x3f400038  
A10     : 0x00000001  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x3ffc7cac  
A14     : 0x00060021  A15     : 0x00060423  SAR     : 0x0000001c  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  
Core 1 was running in ISR context:
EPC1    : 0x400890fd  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400819ef

Backtrace: 0x400819ef:0x3ffbe750 0x40081ccf:0x3ffbe780 0x40083659:0x3ffbe7b0 0x40086e4d:0x3ffbe7d0 0x400890fa:0x3ffba190 0x40085267:0x3ffba1b0 0x4008b259:0x3ffba1d0

Rebooting...
`

fdivitto, 2020-07-12:
please add details, when it happens? Which demo?

KrzysztofJeszke, 2020-07-12:
It happens everywhere, no matter if it's a demo or not. It gets the kernel panic a few times before showing anything on VGA.

fdivitto, 2020-07-12:
Which ESP32 are you using (board, revision...)?

KrzysztofJeszke, 2020-07-12:
I'm using an ESP32-DevKitC-32D with the ESP-WROOM-32D SoC. The board revision is v4.

fdivitto, 2020-07-12:
I'm sorry, I don't have it. 
Looking at the ESP-WROOM-32D documentation I don't see any gpio compatiblity problem with the default settings of fabgl, so the only way I have to look into this issue is to have one, so I've just ordered it.

fdivitto, 2020-07-12:
In the meantime, you could help me! In Arduino IDE you may install the  [ESP Exception Decoder](https://github.com/me-no-dev/EspExceptionDecoder), compile a demo, run it, and paste the exception into the Decoder plugin and post here what the decoder shows.

KrzysztofJeszke, 2020-07-12:
[https://hatebin.com/edmllizcws](url), here you go.

fdivitto, 2020-07-12:
Thank you! Just as comparison, this is Altair 8800 example. Could you try ModelineStudio example?

KrzysztofJeszke, 2020-07-13:
Alright, the ModelineStudio example didn't crash nor spit out any kernel panics.

fdivitto, 2020-07-13:
Ok, this is a SPIFFS and Flash access race. It should has been fixed with one of the last commits. Please replace the version you have installed (examples and fabgl library) with code you find online here.

closed 2020-07-25 by fdivitto