fdivitto/FabGL #69: Serial Interface on TTGO-VGA32 V1.2 board

MartinHepperle, 2020-07-28:
I use a TTGO VGA32 board which exposes IO2 and GPIO12 for a serial TTL interface for a serial terminal emulator.
A common TTL-RS-232C converter (MAX 232) is connected to these pins.

The ANSITerminal firmware does not start as along as something is connected to the GPIO2 pin - only when I disconnect this pin, the firmware boots. After the display of the startup messages, I can re-connect the RX/TX pins and everything works. 
I could not find a reason for this behavior. 
Could there be there some interference of GPIO2 and 12 with the flash memory upload connection via USB?

[BTW: I subclassed the Terminal class and added a Tektronix vector graphics terminal emulation to the ANSI emulation.]

MartinHepperle, 2020-08-01:
... the boot process is also blocked, when the pin GPIO12 is pulled down or up - only when it is  free floating the ESP32 starts...

fdivitto, 2020-08-01:
GPIO12 and GPIO2 usage is documented in AnsiTerminal.ino source code (and of course in ESP32 official docs).

GPIO2: must be either left unconnected/floating, or driven Low, in order to enter the serial bootloader. In normal boot mode (GPIO0 high), GPIO2 is ignored.

GPIO12: It selects flash voltage. To use it disable GPIO12 detection setting efuses with:
                   python espefuse.py --port /dev/cu.SLAB_USBtoUART set_flash_voltage 3.3V
WARN!! Good for ESP32 with 3.3V voltage (ie ESP-WROOM-32). This will BRICK your ESP32 if the flash isn't 3.3V

MartinHepperle, 2020-08-03:
Thank you for your comments.
I solved the problem by not using GPIO 12 (floats to select 3.3V) and instead I soldered a RX wire to GPIO 14. So I use GPIO 2 and 14 connected to the MAX 232 converter. 
This works perfectly for me.

I don't know why they chose to link GPIO 12 to the 4-pin header on the TTGO VGA32 V 1.2 board - the V 1.4 board seems to offer GPIO 14 too.

BTW: thank you for your great library!

closed 2020-10-10 by fdivitto