fdivitto/FabGL #74: TTGO VGA32 V1.4

nippur72, 2020-08-05:
(I tried to add this info on the Wiki but it's read only)

TTGO now ships the updated V1.4 version of their board which is FabGL compatible except for 2 of the SD card pins: `MOSI` and `MISO` are `12` and `2` instead of `16` and `17`. 

Regarding this, would be possible to have the GPIO pin assignments in a gloabal header file e.g. `GPIO.h` so it's easier to change them in one single place?

[image]

panggihsieh, 2020-09-10:
Is it possible to add the x-y-sw joystick to the reserved pinout 2,12,13,14,34,39 in the VIC20 example file?

fdivitto, 2020-09-17:
Last commit adapts default GPIOs for SD card when TTGO board is detected.

closed 2020-09-17 by fdivitto

WacKEDmaN, 2020-10-16:
i just got one of these.... i thought it was quite interesting that the Audio.ino from FabGL is preloaded on it