fdivitto/FabGL #81: SimpleTerminalOut resolution issue

WacKEDmaN, 2020-10-14:
it seems i can only output resolutions below 640x350@70hz with this ino..
the default code has it set for 640x480@60Hz.. but it just has a black screen, even tho the code is running, it doesnt error, ive added serial debug checks to the code and its all running sweet..with plenty of memory being reported (Demo 1 Start: Free Memory: 4374804 bytes)
the 640x480@60hz resolution works fine from modelinestudio.ino aswell as DirectVGA.ino.. 
what am i missing! 
im on a TTGO WROVER-B with 8M PSRAM and default pinout...

fdivitto, 2020-10-14:
is your pinout with 8 or 64 colors?

WacKEDmaN, 2020-10-14:
how do i tell?! im just running the default pin assignment... i have 16 colors showing in the terminal....  
```
 HSync => 23
  VSync => 15
  R0    => 22
  R1    => 21
  G0    => 19
  G1    => 18
  B0    => 5
  B1    => 4
```

fdivitto, 2020-10-14:
ok, this is the 64 colors configuration. Please post a screenshot of your board.

WacKEDmaN, 2020-10-14:
just a TTGO  T18v3.0 with vga connected through 400 and 800ohm resistors..
[image]

fdivitto, 2020-10-14:
> how do i tell?! im just running the default pin assignment... i have 16 colors showing in the terminal....

the default assignment (calling begin() without parameters) is:

HSync => 23
VSync => 15
R0 => 21
R1 => 22
G0 => 18
G1 => 19
B0 => 4
B1 => 5

The difference is in RGB channels bit order, but this should not show a blank screen.
Anyway please, just as test, replace in SimpleTerminalOut.ino, this:

`  DisplayController.begin();`

with this

`  DisplayController.begin(GPIO_NUM_21, GPIO_NUM_22, GPIO_NUM_18, GPIO_NUM_19, GPIO_NUM_4, GPIO_NUM_5, GPIO_NUM_23, GPIO_NUM_15);`

fdivitto, 2020-10-14:
just for test, could you disable PSRAM and then upload the app?

WacKEDmaN, 2020-10-14:
bingo!.. PSRAM seems to be the issue.. disabled it and its now working fine... 
Free Memory: 151788bytes
but that makes me wonder why this one is not working with PSRAM...when all the others do just fine...

fdivitto, 2020-10-14:
ok, this is a good news, now I know where to search. Let me find a solution...

fdivitto, 2020-10-14:
please test last commit: it should solve this issue.

Thanks

WacKEDmaN, 2020-10-14:
that seems to work great with PSRAM enabled now... thanks Fabs
640x480@60hz free memory 4337600bytes :)

if anyone is interested in the debug changes i made to SimpleTerminalOut.ino here it is!...
[SimpleTerminalOut.ino.txt](https://github.com/fdivitto/FabGL/files/5381467/SimpleTerminalOut.ino.txt)

closed 2020-10-15 by WacKEDmaN