fdivitto/FabGL #82: CP/M3 example missing off recent builds

riwide, 2020-10-14:
The CPM/3 example is missing. Whilst it is excellent to have the new multitasking CPM3 workalike,it is not 100% compatible .
It would be good to have the genuine CPM/3 back and supported by the latest codebuilds

fdivitto, 2020-10-14:
I have made all efforts to make it 100% CP/M 3 compatible (please report the applications that don't work). Anyway you may use the Altair8800 example which has a version of CP/M 3 embedded.

closed 2020-10-14 by fdivitto

riwide, 2020-10-14:
Thanks for the response. Sorry for what seems to be a false alarm.I was
having problems with  DEVICE command in your multitasker. Will try again
tomorrow, and try various serial combinations.

Regards
Richard

riwide, 2020-10-14:
I haven't found app problems yet, had problems  getting to cpm3 help,
as name conflicts with your multitasker help.
I will find the real cpm3 and give that a whirl.

Regards
Richard

fdivitto, 2020-10-14:
I don't exclude it is a bug, but main goal is to make this app CP/M 3 compatible. As normal CP/M, to differentiate internal commands, just add .com (help.com).