fdivitto/FabGL #84: mouse pointer issue after enabling wifi

WacKEDmaN, 2020-10-16:
ive just tested both the VIC-20 and HardwareTest... as soon as wifi is enabled the mouse pointer disappears!
keyboard still works fine but no mouse! (tested with psram both on and off!)

fdivitto, 2020-10-28:
I cannot replicate. Please add details about your board.

WacKEDmaN, 2020-10-28:
TTGO ESP32-VGA v1.4

fdivitto, 2020-10-28:
just tested a TTGO ESP32-VGA 1.4, same result, mouse works correctly even when WiFi is enabled. I've tried  it using the internal Flash or inserting an SD card.

Are you using the lastest snapshot on github?

Also, please let me know which board have you selected in Boards, and the parameters you are using )...I mean this menu:
[image]

WacKEDmaN, 2020-10-28:
interesting, ive grabbed latest snapshot and tried vic-20.. and yes mouse still disappears when turning Wi-Fi on...
Boards settings are the same except for 3MB APP partition..
[image]

fdivitto, 2020-10-28:
this is a PSRAM related problem. Try to disable it...
I hope to find a solution!

WacKEDmaN, 2020-10-28:
unfortunately its still doing it with PSRAM disabled... 
thanks Fabs!

fdivitto, 2020-10-29:
> unfortunately its still doing it with PSRAM disabled...
> thanks Fabs!

perfect, so this is not replicable for me...and maybe you have a faulty board.

fdivitto, 2020-10-29:
To simplify take just HardwareTest example, this is the situation (at least what I have with TTGO v.14 board):

- compiling and uploading with PSRAM enabled:
   * moving mouse the screen flicks
   * enabling WiFi, it works, but screen still flicks whenever mouse moves

- compiling and uploading with PSRAM disabled:
   * no problem with WiFi enabled and disabled

Unfortunately I can only fix what I'm able to replicate.

WacKEDmaN, 2020-10-29:
very interesting... i dont get the screen flickering from moving the mouse, it only flickers when wifi is communicating...(the mouse wont work with wifi at all)..and no difference between PSRAM enabled or disabled...
im wondering if its the mouse itself..i dont have another to test with tho :/

fdivitto, 2020-10-29:
HardwareTest just checks for available wifi networks, so you mean that screen flicks just when you press wifi "Test" button and it search for wifi networks, after that screen doesn't flick? But, after that, the mouse stops. Right?

WacKEDmaN, 2020-10-29:
yes thats right

fdivitto, 2020-10-29:
can you do a test for me? In HardwareTest, setup() function, please insert following lines:

```
void setup()
{
  Serial.begin(115200); delay(500); Serial.write("\n\n\n"); // DEBUG ONLY
  Serial.printf("WIFI_TASK_CORE_ID=%d\n", WIFI_TASK_CORE_ID);
  Serial.printf("FABGLIB_VIDEO_CPUINTENSIVE_TASKS_CORE=%d\n", FABGLIB_VIDEO_CPUINTENSIVE_TASKS_CORE);

```

....and of course in the serial monitor, let me know the shown values. Thanks!

WacKEDmaN, 2020-10-29:
WIFI_TASK_CORE_ID=0
FABGLIB_VIDEO_CPUINTENSIVE_TASKS_CORE=1

that looks fine... 

im gonna try it on another power source... maybe it cant power the mouse and wifi with PCs USB supply (even tho its a new PC!)

closed 2020-11-01 by WacKEDmaN