fdivitto/FabGL #85: How to use FFAT with VGADirectController?

abelykh0, 2020-10-19:
Tools > Board > ESP32 Arduino > ESP32 Dev Module
Tools > Partition Scheme > Default 4MB with ffat (1.2 MB APP/1.5 MB FATFS)
Sketch > Include Library > Add .ZIP Library > FabGL-master.zip (that’s because I am using latest master, not v0.9.0)
Use the following source code (slightly modified FabGL/examples/VGA/DirectVGA/DirectVGA.ino):
Upload any files on ffat (see instructions here: https://github.com/me-no-dev/arduino-esp32fs-plugin)

```c++
#include "fabgl.h"
#include "FS.h"
#include "FFat.h"

void listDir(fs::FS &fs, const char * dirname, uint8_t levels){
    Serial.printf("Listing directory: %s\r\n", dirname);

    File root = fs.open(dirname);
    if(!root){
        Serial.println("- failed to open directory");
        return;
    }
    if(!root.isDirectory()){
        Serial.println(" - not a directory");
        return;
    }

    File file = root.openNextFile();
    while(file){
        if(file.isDirectory()){
            Serial.print("  DIR : ");
            Serial.println(file.name());
            if(levels){
                listDir(fs, file.name(), levels -1);
            }
        } else {
            Serial.print("  FILE: ");
            Serial.print(file.name());
            Serial.print("\tSIZE: ");
            Serial.println(file.size());
        }
        file = root.openNextFile();
    }
}

volatile double      objX    = 300;
volatile double      objY    = 200;
static double        objDir  = 0.3;
static double        objVel  =  10;
static constexpr int objSize =  25;
static TaskHandle_t  mainTaskHandle;
static uint32_t      bgParkMillerState = 1;
static constexpr int borderSize = 20;

// just to avoid floating point calculations inside MyDirectDrawVGAController::drawScanline()
volatile int objIntX;
volatile int objIntY;

struct MyDirectDrawVGAController : public fabgl::VGADirectController {

  void IRAM_ATTR drawScanline(uint8_t * dest, int scanLine);

} DisplayController;

inline int fastRandom()
{
  bgParkMillerState = (uint64_t)bgParkMillerState * 48271 % 0x7fffffff;
  return bgParkMillerState % 4;
}

void IRAM_ATTR MyDirectDrawVGAController::drawScanline(uint8_t * dest, int scanLine)
{
  auto fgcolor = createRawPixel(RGB222(3, 0, 0)); // red
  auto bgcolor = createRawPixel(RGB222(0, 0, 2)); // blue

  // fill upper and lower border with random background color
  if (scanLine < borderSize || scanLine > getScreenHeight() - borderSize)
    bgcolor = createRawPixel(RGB222(fastRandom(), fastRandom(), fastRandom()));

  auto width = getScreenWidth();

  // fill line with background color
  memset(dest, bgcolor, width);

  // fill object with foreground color
  if (scanLine >= objIntY - objSize / 2 && scanLine <= objIntY + objSize / 2) {
    for (int col = objIntX - objSize / 2; col < objIntX + objSize / 2; ++col) {
      if (col >= 0 && col < width) {
        VGA_PIXELINROW(dest, col) = fgcolor;
      }
    }
  }

  if (scanLine == getScreenHeight() - 1) {
    // signal end of screen
    vTaskNotifyGiveFromISR(mainTaskHandle, NULL);
  }
}

void setup()
{
  //Serial.begin(115200); delay(500); Serial.write("\n\n\n"); // DEBUG ONLY

  mainTaskHandle = xTaskGetCurrentTaskHandle();

  DisplayController.begin();
  DisplayController.setResolution(VGA_640x480_60Hz);

    if(!FFat.begin()){
        Serial.println("FFat Mount Failed");
        return;
    }

    listDir(FFat, "/", 0);
}

void loop()
{
}
```

```
Result:
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Core 1 register dump:
PC      : 0x40082b09  PS      : 0x00050034  A0      : 0x40085e88  A1      : 0x3ffbe7b0
A2      : 0x3ffc046c  A3      : 0x3ff6d000  A4      : 0x3ffc07a8  A5      : 0x4008c1e8
A6      : 0x00000000  A7      : 0xffffffff  A8      : 0x3ffc0470  A9      : 0xbad00bad
A10     : 0x3ffc046c  A11     : 0x3ffb8abc  A12     : 0x0000000d  A13     : 0x3ffb1cb0
A14     : 0x00000000  A15     : 0x3ffb1cec  SAR     : 0x0000001a  EXCCAUSE: 0x00000007
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000
Core 1 was running in ISR context:
EPC1    : 0x4008f58e  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40082b09

Backtrace: 0x40082b09:0x3ffbe7b0 0x40085e85:0x3ffbe7d0 0x4008f58b:0x3ffb1d10 0x4008fe1d:0x3ffb1d30 0x400886b8:0x3ffb1d50 0x400f6fba:0x3ffb1dc0 0x400f508e:0x3ffb1df0 0x400f4439:0x3ffb1e10 0x400f398b:0x3ffb1ea0 0x400dce49:0x3ffb1f10 0x400d3b0c:0x3ffb1f50 0x400d147a:0x3ffb1f90 0x400d458f:0x3ffb1fb0 0x40089c71:0x3ffb1fd0
```

abelykh0, 2020-10-19:
BTW, if I comment line 
```c++
//DisplayController.setResolution(VGA_640x480_60Hz);
```
the `listDir` works fine.

fdivitto, 2020-10-28:
please test last commit. In my tests it should fix this issue.

abelykh0, 2020-10-29:
Thanks, perfect!

abelykh0, 2020-11-03:
Hello @fdivitto,

I have the same issue occurring randomly in a more complex scenario (Spectrum ZX emulator). Would you have time to take a look or give me a hint on where to look?

Andrey

fdivitto, 2020-11-28:
because MyDirectDrawVGAController::drawScanline is called inside an interrupt with IRAM flag enabled, you should make sure that all your calls (even depth function calls) are stored in IRAM. If you have some code I can test (without external dependences for simplicity) I can help to fix this issue.

abelykh0, 2020-11-29:
I looked through the code and it seems that all methods that I call have IRAM_ATTR. 

As I said, unfortunately it is not as easy to replicate as before your fix. I created a branch "test" https://github.com/abelykh0/esp32-z80emu/tree/test. If you have a chance, please take a quick look. I am using Visual Studio Code with PlatformIO. Before building, copy your latest source code under `/lib`. Also, you need to load files under `/data` unto FatFS on flash.

To replicate, select F3, then use arrows, at random point it crashes. 

I am really impressed with your library and just want it to become even better :)

fdivitto, 2020-11-29:
Please help me to test your project: I can compile and upload it using VS Code with PlatformIO. Monitor says:
```
in setup()
[E][FFat.cpp:60] begin(): Mounting FFat partition failed! Error: -1
FFat Mount Failed
....
```

I've tried to "Build Filesystem Image" and "Upload Filesystem Image", but a red "FAILED" is shown. I have to install something other to upload the ffat filesystem?

Sorry, but I'm not very familiar with platformio.

abelykh0, 2020-11-30:
Unfortunately, there's no option for this in PlatformIO.

I think the easiest is to use this tool: https://github.com/lorol/arduino-esp32fs-plugin. 

After installing the tool, in Arduino IDE:
- open  `esp32-z80emu.ino` (just a dummy empty file, I added it to the project)
- select Tools > Partition Scheme > "Default 4MB with ffat (1.2MB App / 1.5MB FatFS)"
- then Tools > ESP32 Sketch Data Upload, then "FatFS", OK

Just in case I added the files from this tool into `/ESP32FS` as well as empty `esp32-z80emu.ino` to the `test` branch. The `/ESP32FS` should be copied under `C:\Program Files (x86)\Arduino\tools`, then restart Arduino IDE.

fdivitto, 2020-11-30:
Thank you, now I can replicate.

fdivitto, 2020-11-30:
I've to investigate better, but it seems that vtables are stored in Flash, not in RAM. This is a problem when flash is unaccessible (due fatfs) and a scanline interrupt occurs.

fdivitto, 2020-11-30:
This is very sad, we cannot use virtual and abstract methods while drawing a scanline.
Just for test, I've implemented scanline drawing as a callback, instead of virtual method.

Please download "vgadirectcontroller.h" and "vgadirectcontroller.cpp" from here:
[files to replace](https://drive.google.com/drive/folders/1Tn2B-MzDiPlQZH3KBNi5bSg8IDedePq2?usp=sharing)
...and copy into fabgl folder.

Now you have to make ugly changes in your code. For example:
VideoController::drawScanline must be: 
`static void IRAM_ATTR drawScanline(void* arg, uint8_t* dest, int scanLine);`
Inside VideoController constructor you have to specify the callback:
`setDrawScanlineCallback(drawScanline, this);`
And VideoController::drawScanline must be something like this:
```
void IRAM_ATTR VideoController::drawScanline(void* arg, uint8_t* dest, int scanLine)
{
    auto ths = static_cast<VideoController*>(arg);
    if (scanLine >= ths->_band2Start)
    {
        ths->_band2->drawScanline(dest, scanLine);
    }
    else
    {
        ths->_band1->drawScanline(dest, scanLine);
    }
}
```

Of course you have to also change drawScanline of Band, that cannot no longer be abstract!

Please let me know your results once finished changes.

abelykh0, 2020-12-01:
Thank you for the update. I will need some time to refactor everything, will update tomorrow.

It looks like issue https://github.com/espressif/esp-idf/issues/3434. As you said, if using Arduino, there's no way to force vtable into IRAM.

fdivitto, 2020-12-02:
Thank you for the tests. Changes implemented.

Please let me know when you finish this project: I am very interested!

closed 2020-12-02 by fdivitto

abelykh0, 2020-12-02:
> Thank you for the tests. Changes implemented.
Thank you!

> Please let me know when you finish this project: I am very interested!
Since it is a hobby project, it will be never "finished". As of now, it can emulate Sinclair 128K computer which was popular in the 1980s, especially in UK. I am not really interested to make it 100% correct at the moment (it is currently slightly faster than it is supposed to be and I didn't care about advanced border effects). One of the features that I may want to implement in the near future is "beeper". Basically, every 1/50 of second the Sinclair may set/reset certain pin to create sound. For now, I only partially emulate AY3-8912, but not "beeper".

There are literally thousands of games and various applications available, you can find them here: https://spectrumcomputing.co.uk/.

Also, take a look at a very similar project here: https://github.com/rampa069/ZX-ESPectrum.