fdivitto/FabGL #86: PS2 mouse and keyboard influence each other

racedeyu, 2020-10-21:
This problem can be reproduced with following code. Please run the code, press and hold any key on keyboard, move the mouse quickly and do not press the mouse. 

```cpp
#include <fabgl.h>

fabgl::PS2Controller PS2Controller;

void setup() {
  Serial.begin(115200);
  Serial.printf("START\n");
  PS2Controller.begin(PS2Preset::KeyboardPort0_MousePort1);
  auto hostMouse = PS2Controller.mouse();
  auto hostKeyboard = PS2Controller.keyboard();
  hostKeyboard->setTypematicRateAndDelay(33, 500);
  hostMouse->setSampleRate(200);
}

void loop()
{

  auto hostMouse = PS2Controller.mouse();
  auto hostKeyboard = PS2Controller.keyboard();

  for (;;) {
    if (hostKeyboard->virtualKeyAvailable()) {
      hostKeyboard->getNextVirtualKey(nullptr, 0);
    } else {
      break;
    }
  }

  for (;;) {
    if (hostMouse->deltaAvailable()) {
      MouseDelta mouseDelta;
      hostMouse->getNextDelta(&mouseDelta);
      if (mouseDelta.buttons.left) {
        Serial.printf("BUG!\n");
      }
    } else {
      break;
    }
  }
}
```

fdivitto, 2020-10-21:
You know, the two PS/2 devices are handled by the ULP processor. It is clocked at 8Mhz, but each instruction requires about 6 cycles (that is about 1.3M instructions per second). It is also suspended whenever main CPU accesses the same memory area (it happens very often to get typed keys or mouse movements). The ULP processor can fail sometimes to decode inputs when they occur at the same time. If you set typematic rate at 33ms (maximum speed) and mouse sample rate at 200 samples per second ....then you're making things even more critical!

However I've just added an additional check on parity (I've observed that when there is a clash the parity is wrong): now the behaviour is that if the parity is wrong then the mouse packet decoding is aborted.

Probably someone better than me could get more from the ULP, but for now this is the best we have.

BTW: You should check return value of getNextDelta(). It returns False on error:
```
      if (hostMouse->getNextDelta(&mouseDelta)) {
        if (mouseDelta.buttons.left) {
          Serial.printf("BUG!\n");
        }
      }
```

racedeyu, 2020-10-22:
@fdivitto, thanks for your great work. I've an idea to resolve this problem, we may use ps2 inhibit to handle transmission timing. 

Data Receive:
Once we received a byte from device pull clk  low after stop bit and trig interrupt, after SLOW MEM read out and interrupt processed release clk. 

Data Send:
Copying data to SLOW MEM must be done in RTC interrupt, ULP pull clk low and trig interrupt periodically. Once new data found in interrupt set port mode, fill data to send.

However, This is a big change, handle both two port make things more complicated. I trying to resolve the problems use this way.

fdivitto, 2020-10-22:
no so big change actually, but it doesn't solve the case when both PS/2 devices are sending data at the same time. If you lose first byte of a mouse packet, you lose the sync at packet level.

Have you tried last commit?

racedeyu, 2020-10-22:
If two devices send data at the same time, the port that has sent data complete and enter inhibition first, wait for other port has completed the transmission and enter inhibition and then triggers the interrupt. If just one port received data both port enter inhibition.

 Sorry, DEV board not near by, I will try last commit later.

fdivitto, 2020-10-22:
understand. Anyway before face a change in ULP code I prefer to check if last commit workarounds this issue.

racedeyu, 2020-10-22:
The problem doesn't get better with last commit.

fdivitto, 2020-10-22:
are you sure? I've never got a "bug" with last commit. Make sure you are actually using it.

racedeyu, 2020-10-22:
Maybe I did not compile cleanly before. I had done a clean compilation and added a simple version confirmation code. The probability of the problem is reduced, but it still exists. 

```cpp
#include <fabgl.h>

fabgl::PS2Controller PS2Controller;

void setup() {
  Serial.begin(115200);
  Serial.printf("START\n");
  PS2Controller.begin(PS2Preset::KeyboardPort0_MousePort1);
  auto hostMouse = PS2Controller.mouse();
  auto hostKeyboard = PS2Controller.keyboard();
  hostKeyboard->setTypematicRateAndDelay(33, 500);
  hostMouse->setSampleRate(200);
  if (!PS2Controller.parityError(0)){
    Serial.printf("Version fine\n");
  }
}

void loop()
{

  auto hostMouse = PS2Controller.mouse();
  auto hostKeyboard = PS2Controller.keyboard();

  for (;;) {
    if (hostKeyboard->virtualKeyAvailable()) {
      hostKeyboard->getNextVirtualKey(nullptr, 0);
    } else {
      break;
    }
  }
  
  for (;;) {
    if (hostMouse->deltaAvailable()) {
      MouseDelta mouseDelta;
      hostMouse->getNextDelta(&mouseDelta);
      if (mouseDelta.buttons.left) {
        Serial.printf("BUG!\n");
      }
    } else {
      break;
    }
  }
}
```

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
START
Version fine
BUG!
BUG!
BUG!
BUG!
```

racedeyu, 2020-10-22:
Maybe I did not compile cleanly before. I did a clean compilation and added a simple version confirmation code. The probability of the problem is reduced, but it still exists.

```cpp
#include <fabgl.h>

fabgl::PS2Controller PS2Controller;

void setup() {
  Serial.begin(115200);
  Serial.printf("START\n");
  PS2Controller.begin(PS2Preset::KeyboardPort0_MousePort1);
  auto hostMouse = PS2Controller.mouse();
  auto hostKeyboard = PS2Controller.keyboard();
  hostKeyboard->setTypematicRateAndDelay(33, 500);
  hostMouse->setSampleRate(200);
  if (!PS2Controller.parityError(0)){
    Serial.printf("Version fine\n");
  }
}

void loop()
{

  auto hostMouse = PS2Controller.mouse();
  auto hostKeyboard = PS2Controller.keyboard();

  for (;;) {
    if (hostKeyboard->virtualKeyAvailable()) {
      hostKeyboard->getNextVirtualKey(nullptr, 0);
    } else {
      break;
    }
  }
  
  for (;;) {
    if (hostMouse->deltaAvailable()) {
      MouseDelta mouseDelta;
      if (hostMouse->getNextDelta(&mouseDelta)){
        if (mouseDelta.buttons.left) {
          Serial.printf("BUG!\n");
        }
      }
    } else {
      break;
    }
  }
}
```

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
START
Version fine
BUG!
BUG!
```

racedeyu, 2020-10-22:
The error rate in the real application is acceptable and much better than before, changing the keyboard led when mouse moving was still not good enough. However, it is enough for most applications.

fdivitto, 2020-10-22:
thanks to your previous comments and tips I still have to do a little change that if it works should change things a lot...but I need more time to do it.

fdivitto, 2021-03-11:
I just re-wrote PS/2 (ULP and other stuff) code from scratch. Now it should avoid any clash between mouse and keyboard data streams.
Of course these changes are still "experimental"...so I ask help for testing, Thanks!

guidol70, 2021-03-16:
@fdivitto the old CHERRY Keyboard **doesnt lockup** in the Keyboard-Studio!

racedeyu, 2021-03-16:
@fdivitto Apologies for my delay, I missed the email, I will test it soon.

racedeyu, 2021-03-16:
I still need time to prepare the test environment level to be familiar with the code, but I used a logic analyzer to capture some waveforms and hope it helps.

The "Break" Key waveform, which took 72ms in total. 
[image]

This is the first byte of "Break". The small pulse (mark 1) should not exist, and clk should be pulled low by the Host before the falling edge of the stop bit. The mouse only enters the inhibit (mark 2) in the data field of the keyboard. I don't know if this is designed to do so or a defect. 
[image]

This is the first byte of "Break" sent when the mouse is moving. 
[image]

racedeyu, 2021-03-16:
In the tx sequence, Clk should be pulled low for at least 100us and then Dat pulled low. After stop bit, there is a small pulse that shouldn't be there. 
[image]

racedeyu, 2021-03-16:
After reading the code and doing a simple test, no problems were found in the function. But the timing is different from spec.

“The mouse only enters the inhibit (mark 2) in the data field of the keyboard.” That is ok, same with the design.

fdivitto, 2021-03-16:
Thanks for the tests!

> This is the first byte of "Break". The small pulse (mark 1) should not exist, and clk should be pulled low by the Host before the falling edge of the stop bit. The mouse only enters the inhibit [...]

CLK should have eleven falling edges controlled by the keyboard. After it host drive CLK low, to avoid keyboard (or mouse) to send other data. This is the reason of the small pulse, which is not actually a pulse, but a "drive low".

Receiving stall should be caused by ULP that sometimes cannot generate interrupt at the end of frame ("wake" ULP instruction). This happens when RTC is not ready, but we cannot know when RTC is ready because ULP and SoC are continuously in running state.

I have posted a commit which should workaround this problem.

fdivitto, 2021-03-16:
> In the tx sequence, Clk should be pulled low for at least 100us and then Dat pulled low. After stop bit, there is a small pulse that shouldn't be there.

CLK and DAT are pulled low for 200us, at the same time. I have not found in IBM documentation that there must be a difference (clk down, pause, dat down).
As you can see from documentation last drive down in DAT is produced by the keyboard (point 7).

[image]

fdivitto, 2021-03-16:
> update:
> When I moved the mouse, I pressed Numlock to change the LED of the keyboard, and found that the movement of the mouse and the keys of mouse are still be interfered by the keyboard. I need to construct code for testing, and capture the waveform of that situation.

the "big change" from the past in PS/2 routines is that now while receiving on port 0, port 1 is disabled (CLK low by the host) and viceversa. The same happens on sending. Nothing can happen on the other port during TX or RX. Also, now  start and stop bits are checked, and finally when mouse is out of sync is also detected (at mouse frame level).
I've tested your initial testing code consuming my mouse and keyboard, and I have never got a "BUG" :-)

racedeyu, 2021-03-17:
> CLK should have eleven falling edges controlled by the keyboard. After it host drive CLK low, to avoid keyboard (or mouse) to send other data. This is the reason of the small pulse, which is not actually a pulse, but a "drive low".

The host should drive CLK low after 11 falling edges, and before the subsequent rising edge (device and host drive CLK low at the same time), as point 5 and timing diagram.

[image]

> CLK and DAT are pulled low for 200us, at the same time. I have not found in IBM documentation that there must be a difference (clk down, pause, dat down).

This is a capture of the timing of the communication between my computer and the keyboard. CLK pulled low first, and then DAT pulled low after a period of time. This may be related to the point 3 in the figure above. 

[image]

fdivitto, 2021-03-17:
have you tested last commit?

racedeyu, 2021-03-17:
> 
> 
> have you tested last commit?

I have already tested before the last post, but I don’t know what kind of working condition the last commit aimed at.

Still the ”Break“ key.
[image]

fdivitto, 2021-03-17:
> Still the ”Break“ key.
> [image]

0xe1 0x14 0x77 ... is PAUSE, not BREAK (you know, Break is done pressing CTRL-PAUSE).
The spike after stop bit is just the moment when keyboard releases (going UP) CLK and ESP32 brings it LOW, so I cannot see nothing of wrong in it.

If there are still problems with keyboard and/or mouse please help me to replicate on my side (with a demo app, ....)

racedeyu, 2021-03-17:
>0xe1 0x14 0x77 ... is PAUSE, not BREAK (you know, Break is done pressing CTRL-PAUSE).
I made a low-level mistake. I tried to use my keyboard with previous commit to reproduce #105 , but I couldn't get any wrong. And now PS2 is functionally perfect, this issue has been solved, thank you.

closed 2021-03-17 by fdivitto