fdivitto/FabGL #94: Missing /src/emudevs when install FabGL 1.0.0 library in Arduino-IDE

guidol70, 2021-01-22:
Today I did get my TTGO VGA32 v1.4 and would like to compile/install the Multitasking-CPM.

While compiling the Arduino-IDE (v1.8.3) did miss the Z80.h in /src/emudevs :(
I had installed the FabGL library in the IDE some days ago (before my board did arrive).

I checked the library-directory and it isnt there :(
So I checked against the github-content and found there the /src/emudevs directory.
Copied (/src/emudevs) into the library directory of the IDE and the sketch did compile ( at the 2nd try - why will write above)
Path C:\Users\Username\Documents\Arduino\libraries\FabGL

To recheck I deleted the FabGL library from the IDE and did reinstall FabGL library 1.0.0 - and again the /src/emudevs directory is missing.

Isnt the library not directly installed from your github-directory? So maybe the library "cache" of the Arduino IDE isnt complete?

As the 2nd compile worked:
There was a "space" problem for the sketch in my device, because I had changed my platform.txt (for a RunCPM-project) to optimize the compile for speed and not for size (compile option -O3 = speed  - normally its set to -Os for size).

After reverting the compile option and restarting the IDE (because it seem to cache the compile and had bigger compiled sketch size) I could compile without errors and the emulation did start fine :)

[FabGL_MultitakingCPM_compile_error.txt](https://github.com/fdivitto/FabGL/files/5855658/FabGL_MultitakingCPM_compile_error.txt)

fdivitto, 2021-01-23:
thank you!
I've just released 1.0.1, it should fix these problems.

closed 2021-01-23 by fdivitto

guidol70, 2021-01-23:
No problem ;)
I re-checked it with the installation of the v1.0.1-library version and the directory /src/emudevs is now included :)
Many Thanks for that and the fine emulations!