fdivitto/FabGL #95: compilation errors with 1.0.1

WacKEDmaN, 2021-01-26:
i cant use this lib at all with the latest version... everything errors out on compile...
eg MultitaskingCPM
```
In file included from E:\WacKEDmaN\Documents\Arduino\libraries\FabGL-1.0.1\src\dispdrivers\vgadirectcontroller.cpp:41:0:
E:\WacKEDmaN\Documents\Arduino\libraries\FabGL-1.0.1\src\dispdrivers\vgadirectcontroller.h: In constructor 'fabgl::VGADirectController::VGADirectController(bool)':
E:\WacKEDmaN\Documents\Arduino\libraries\FabGL-1.0.1\src\dispdrivers\vgadirectcontroller.h:219:32: error: 'fabgl::VGADirectController::m_autoRun' will be initialized after [-Werror=reorder]
   bool                         m_autoRun;
                                ^
E:\WacKEDmaN\Documents\Arduino\libraries\FabGL-1.0.1\src\dispdrivers\vgadirectcontroller.h:211:32: error:   'int32_t fabgl::VGADirectController::m_linesCount' [-Werror=reorder]
   int32_t                      m_linesCount;
                                ^
E:\WacKEDmaN\Documents\Arduino\libraries\FabGL-1.0.1\src\dispdrivers\vgadirectcontroller.cpp:66:1: error:   when initialized here [-Werror=reorder]
 VGADirectController::VGADirectController(bool autoRun)
 ^
cc1plus.exe: some warnings being treated as errors
Multiple libraries were found for "WiFi.h"
 Used: C:\Users\WacKEDmaN\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.5-rc6\libraries\WiFi
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
exit status 1
Error compiling for board ESP32 Dev Module.

```

Altair8800
```
In file included from E:\WacKEDmaN\Documents\Arduino\libraries\FabGL-1.0.1\src\dispdrivers\vgadirectcontroller.cpp:41:0:
E:\WacKEDmaN\Documents\Arduino\libraries\FabGL-1.0.1\src\dispdrivers\vgadirectcontroller.h: In constructor 'fabgl::VGADirectController::VGADirectController(bool)':
E:\WacKEDmaN\Documents\Arduino\libraries\FabGL-1.0.1\src\dispdrivers\vgadirectcontroller.h:219:32: error: 'fabgl::VGADirectController::m_autoRun' will be initialized after [-Werror=reorder]
   bool                         m_autoRun;
                                ^
E:\WacKEDmaN\Documents\Arduino\libraries\FabGL-1.0.1\src\dispdrivers\vgadirectcontroller.h:211:32: error:   'int32_t fabgl::VGADirectController::m_linesCount' [-Werror=reorder]
   int32_t                      m_linesCount;
                                ^
E:\WacKEDmaN\Documents\Arduino\libraries\FabGL-1.0.1\src\dispdrivers\vgadirectcontroller.cpp:66:1: error:   when initialized here [-Werror=reorder]
 VGADirectController::VGADirectController(bool autoRun)
 ^
cc1plus.exe: some warnings being treated as errors
exit status 1
Error compiling for board ESP32 Dev Module.
```

HardwareTest
```
In file included from E:\WacKEDmaN\Documents\Arduino\libraries\FabGL-1.0.1\src\dispdrivers\vgadirectcontroller.cpp:41:0:
E:\WacKEDmaN\Documents\Arduino\libraries\FabGL-1.0.1\src\dispdrivers\vgadirectcontroller.h: In constructor 'fabgl::VGADirectController::VGADirectController(bool)':
E:\WacKEDmaN\Documents\Arduino\libraries\FabGL-1.0.1\src\dispdrivers\vgadirectcontroller.h:219:32: error: 'fabgl::VGADirectController::m_autoRun' will be initialized after [-Werror=reorder]
   bool                         m_autoRun;
                                ^
E:\WacKEDmaN\Documents\Arduino\libraries\FabGL-1.0.1\src\dispdrivers\vgadirectcontroller.h:211:32: error:   'int32_t fabgl::VGADirectController::m_linesCount' [-Werror=reorder]
   int32_t                      m_linesCount;
                                ^
E:\WacKEDmaN\Documents\Arduino\libraries\FabGL-1.0.1\src\dispdrivers\vgadirectcontroller.cpp:66:1: error:   when initialized here [-Werror=reorder]
 VGADirectController::VGADirectController(bool autoRun)
 ^
cc1plus.exe: some warnings being treated as errors
Multiple libraries were found for "WiFi.h"
 Used: C:\Users\WacKEDmaN\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.5-rc6\libraries\WiFi
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
exit status 1
Error compiling for board ESP32 Dev Module.

```
All examples are showing the same as above....
```
DirectVGA_ElMuro ..is showing a slightly different error...
In file included from E:\WacKEDmaN\Documents\Arduino\libraries\FabGL-1.0.1\examples\VGA\DirectVGA_ElMuro\DirectVGA_ElMuro.ino:30:0:
sketch\controllers.h: In constructor 'GameController::GameController(int, gcmode)':
controllers.h:11:12: error: 'GameController::id' will be initialized after [-Werror=reorder]
   int id = 0;     // device number assigned by client
            ^
controllers.h:10:17: error:   'gcmode GameController::mode' [-Werror=reorder]
   gcmode mode = MODE_DIRECTIONAL;
                 ^
controllers.h:14:3: error:   when initialized here [-Werror=reorder]
   GameController( int id_, gcmode mode_) : id( id_ ), mode(mode_) { }
   ^
In file included from E:\WacKEDmaN\Documents\Arduino\libraries\FabGL-1.0.1\examples\VGA\DirectVGA_ElMuro\DirectVGA_ElMuro.ino:31:0:
sketch\soundchip.h: In function 'void iPlaySound(void*)':
sketch\soundchip.h:78:10: warning: 'pwave' may be used uninitialized in this function [-Wmaybe-uninitialized]
   delete pwave;
          ^
cc1plus.exe: some warnings being treated as errors
exit status 1
'GameController::id' will be initialized after [-Werror=reorder]

```

WacKEDmaN, 2021-01-26:
latest master is throwing errors in keyboard.h....
```
In file included from E:\WacKEDmaN\Documents\Arduino\libraries\FabGL\src\devdrivers\keyboard.cpp:31:0:
E:\WacKEDmaN\Documents\Arduino\libraries\FabGL\src\devdrivers\keyboard.h: In constructor 'fabgl::Keyboard::Keyboard()':
E:\WacKEDmaN\Documents\Arduino\libraries\FabGL\src\devdrivers\keyboard.h:429:29: error: 'fabgl::Keyboard::m_lastDeadKey' will be initialized after [-Werror=reorder]
   VirtualKey                m_lastDeadKey;
                             ^
E:\WacKEDmaN\Documents\Arduino\libraries\FabGL\src\devdrivers\keyboard.h:416:29: error:   'uint8_t fabgl::Keyboard::m_scancodeSet' [-Werror=reorder]
   uint8_t                   m_scancodeSet;
                             ^
E:\WacKEDmaN\Documents\Arduino\libraries\FabGL\src\devdrivers\keyboard.cpp:46:1: error:   when initialized here [-Werror=reorder]
 Keyboard::Keyboard()
 ^
E:\WacKEDmaN\Documents\Arduino\libraries\FabGL\src\devdrivers\keyboard.cpp: In member function 'fabgl::VirtualKey fabgl::Keyboard::scancodeToVK(uint8_t, bool, const fabgl::KeyboardLayout*)':
E:\WacKEDmaN\Documents\Arduino\libraries\FabGL\src\devdrivers\keyboard.cpp:532:12: error: enumeration value 'VK_NONE' not handled in switch [-Werror=switch]
     switch (vk) {
            ^
E:\WacKEDmaN\Documents\Arduino\libraries\FabGL\src\devdrivers\keyboard.cpp:532:12: error: enumeration value 'VK_SPACE' not handled in switch [-Werror=switch]
E:\WacKEDmaN\Documents\Arduino\libraries\FabGL\src\devdrivers\keyboard.cpp:532:12: error: enumeration value 'VK_0' not handled in switch [-Werror=switch]
E:\WacKEDmaN\Documents\Arduino\libraries\FabGL\src\devdrivers\keyboard.cpp:532:12: error: enumeration value 'VK_1' not handled in switch [-Werror=switch]
E:\WacKEDmaN\Documents\Arduino\libraries\FabGL\src\devdrivers\keyboard.cpp:532:12: error: enumeration value 'VK_2' not handled in switch [-Werror=switch]
E:\WacKEDmaN\Documents\Arduino\libraries\FabGL\src\devdrivers\keyboard.cpp:532:12: error: enumeration value 'VK_3' not handled in switch [-Werror=switch]
E:\WacKEDmaN\Documents\Arduino\libraries\FabGL\src\devdrivers\keyboard.cpp:532:12: error: enumeration value 'VK_4' not handled in switch [-Werror=switch]
E:\WacKEDmaN\Documents\Arduino\libraries\FabGL\src\devdrivers\keyboard.cpp:532:12: error: enumeration value 'VK_5' not handled in switch [-Werror=switch]
E:\WacKEDmaN\Documents\Arduino\libraries\FabGL\src\devdrivers\keyboard.cpp:532:12: error: enumeration value 'VK_6' not handled in switch [-Werror=switch]
E:\WacKEDmaN\Documents\Arduino\libraries\FabGL\src\devdrivers\keyboard.cpp:532:12: error: enumeration value 'VK_7' not handled in switch [-Werror=switch]
E:\WacKEDmaN\Documents\Arduino\libraries\FabGL\src\devdrivers\keyboard.cpp:532:12: error: enumeration value 'VK_8' not handled in switch [-Werror=switch]
E:\WacKEDmaN\Documents\Arduino\libraries\FabGL\src\devdrivers\keyboard.cpp:532:12: error: enumeration value 'VK_9' not handled in switch [-Werror=switch]
E:\WacKEDmaN\Documents\Arduino\libraries\FabGL\src\devdrivers\keyboard.cpp:532:12: error: enumeration value 'VK_KP_0' not handled in switch [-Werror=switch]
E:\WacKEDmaN\Documents\Arduino\libraries\FabGL\src\devdrivers\keyboard.cpp:532:12: error: enumeration value 'VK_KP_1' not handled in switch [-Werror=switch]
E:\WacKEDmaN\Documents\Arduino\libraries\FabGL\src\devdrivers\keyboard.cpp:532:12: error: enumeration value 'VK_KP_2' not handled in switch [-Werror=switch]
E:\WacKEDmaN\Documents\Arduino\libraries\FabGL\src\devdrivers\keyboard.cpp:532:12: error: enumeration value 'VK_KP_3' not handled in switch [-Werror=switch]
E:\WacKEDmaN\Documents\Arduino\libraries\FabGL\src\devdrivers\keyboard.cpp:532:12: error: enumeration value 'VK_KP_4' not handled in switch [-Werror=switch]
E:\WacKEDmaN\Documents\Arduino\libraries\FabGL\src\devdrivers\keyboard.cpp:532:12: error: enumeration value 'VK_KP_5' not handled in switch [-Werror=switch]
E:\WacKEDmaN\Documents\Arduino\libraries\FabGL\src\devdrivers\keyboard.cpp:532:12: error: enumeration value 'VK_KP_6' not handled in switch [-Werror=switch]
E:\WacKEDmaN\Documents\Arduino\libraries\FabGL\src\devdrivers\keyboard.cpp:532:12: error: enumeration value 'VK_KP_7' not handled in switch [-Werror=switch]
E:\WacKEDmaN\Documents\Arduino\libraries\FabGL\src\devdrivers\keyboard.cpp:532:12: error: enumeration value 'VK_KP_8' not handled in switch [-Werror=switch]
E:\WacKEDmaN\Documents\Arduino\libraries\FabGL\src\devdrivers\keyboard.cpp:532:12: error: enumeration value 'VK_KP_9' not handled in switch [-Werror=switch]
E:\WacKEDmaN\Documents\Arduino\libraries\FabGL\src\devdrivers\keyboard.cpp:532:12: error: enumeration value 'VK_a' not handled in switch [-Werror=switch]
E:\WacKEDmaN\Documents\Arduino\libraries\FabGL\src\devdrivers\keyboard.cpp:532:12: error: enumeration value 'VK_b' not handled in switch [-Werror=switch]
E:\WacKEDmaN\Documents\Arduino\libraries\FabGL\src\devdrivers\keyboard.cpp:532:12: error: enumeration value 'VK_c' not handled in switch [-Werror=switch]
E:\WacKEDmaN\Documents\Arduino\libraries\FabGL\src\devdrivers\keyboard.cpp:532:12: error: enumeration value 'VK_d' not handled in switch [-Werror=switch]
E:\WacKEDmaN\Documents\Arduino\libraries\FabGL\src\devdrivers\keyboard.cpp:532:12: error: enumeration value 'VK_e' not handled in switch [-Werror=switch]
E:\WacKEDmaN\Documents\Arduino\libraries\FabGL\src\devdrivers\keyboard.cpp:532:12: error: enumeration value 'VK_f' not handled in switch [-Werror=switch]
[... cut ...]
E:\WacKEDmaN\Documents\Arduino\libraries\FabGL\src\devdrivers\keyboard.cpp:532:12: error: enumeration value 'VK_CARET_U' not handled in switch [-Werror=switch]
E:\WacKEDmaN\Documents\Arduino\libraries\FabGL\src\devdrivers\keyboard.cpp:532:12: error: enumeration value 'VK_ASCII' not handled in switch [-Werror=switch]
E:\WacKEDmaN\Documents\Arduino\libraries\FabGL\src\devdrivers\keyboard.cpp:532:12: error: enumeration value 'VK_LAST' not handled in switch [-Werror=switch]
cc1plus.exe: some warnings being treated as errors
exit status 1
Error compiling for board ESP32 Dev Module.

```

WacKEDmaN, 2021-01-26:
both DirectVGA_ElMuro and ClassicRacer both fail after going back to 1.0.0...
```
In file included from E:\WacKEDmaN\Documents\Arduino\libraries\FabGL\examples\VGA\ClassicRacer\ClassicRacer.ino:11:0:
sketch\controllers.h: In constructor 'GameController::GameController(int, gcmode)':
controllers.h:11:12: error: 'GameController::id' will be initialized after [-Werror=reorder]
   int id = 0;     // device number assigned by client
            ^
controllers.h:10:17: error:   'gcmode GameController::mode' [-Werror=reorder]
   gcmode mode = MODE_DIRECTIONAL;
                 ^
controllers.h:14:3: error:   when initialized here [-Werror=reorder]
   GameController( int id_, gcmode mode_) : id( id_ ), mode(mode_) { }
   ^
In file included from E:\WacKEDmaN\Documents\Arduino\libraries\FabGL\examples\VGA\ClassicRacer\ClassicRacer.ino:30:0:
sketch\score.h: In member function 'void Score::drawScores()':
sketch\score.h:176:9: warning: unused variable 'nselected' [-Wunused-variable]
     int nselected = 1;
         ^
In file included from E:\WacKEDmaN\Documents\Arduino\libraries\FabGL\examples\VGA\ClassicRacer\ClassicRacer.ino:30:0:
sketch\score.h: In member function 'virtual void Score::update(int)':
score.h:313:41: error: suggest parentheses around comparison in operand of '==' [-Werror=parentheses]
         canvas.setPenColor(millis()%250 > 125 == 0 ? RGB888(0,255,0):scorecolors[editItem] );  
                                         ^
In file included from E:\WacKEDmaN\Documents\Arduino\libraries\FabGL\examples\VGA\ClassicRacer\ClassicRacer.ino:31:0:
sketch\race.h: In member function 'void Race::winAnimation(int)':
race.h:537:96: error: format '%d' expects argument of type 'int', but argument 3 has type 'long int' [-Werror=format=]
     sprintf( buffer, "TIME : %02d:%02d", (currentTime / 1000) / 60,  (currentTime / 1000) % 60 );
                                                                                                ^
race.h:537:96: error: format '%d' expects argument of type 'int', but argument 4 has type 'long int' [-Werror=format=]
race.h:537:96: error: format '%d' expects argument of type 'int', but argument 3 has type 'long int' [-Werror=format=]
race.h:537:96: error: format '%d' expects argument of type 'int', but argument 4 has type 'long int' [-Werror=format=]
In file included from E:\WacKEDmaN\Documents\Arduino\libraries\FabGL\examples\VGA\ClassicRacer\ClassicRacer.ino:32:0:
sketch\menu.h: At global scope:
sketch\menu.h:23:97: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
   char *textControls[6] = { ".......", " MOUSE ", "JOYSTICK", " ARROWS ", "A S T F ", "Q A O P" };
                                                                                                 ^
sketch\menu.h:23:97: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
sketch\menu.h:23:97: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
sketch\menu.h:23:97: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
sketch\menu.h:23:97: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
sketch\menu.h:23:97: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
In file included from E:\WacKEDmaN\Documents\Arduino\libraries\FabGL\examples\VGA\ClassicRacer\ClassicRacer.ino:29:0:
sketch\soundchip.h: In function 'void iPlaySound(void*)':
sketch\soundchip.h:78:10: warning: 'pwave' may be used uninitialized in this function [-Wmaybe-uninitialized]
   delete pwave;
          ^
cc1plus.exe: some warnings being treated as errors
exit status 1
'GameController::id' will be initialized after [-Werror=reorder]

```

fdivitto, 2021-01-27:
Wow, they are a lot of errors! Please let me know:
- the Arduino version
- in board manager, which ESP32 core library is installed
thank you!

fdivitto, 2021-01-27:
I've just tried a fresh install on MacOS 10.15, Linux Debian 10 and Windows 10, using:
- Arduino version: 1.8.13
- board manager: ESP32 1.0.4-rc1
- fabgl 1.0.1

It compiles correctly all the examples.

fdivitto, 2021-01-27:
ok, I guess you enabled all warnings in Arduino properties and because warnings in Arduino are handled as error...
I'm going to release a version without warnings.

fdivitto, 2021-01-27:
version 1.0.2 fixes this problem

closed 2021-01-27 by fdivitto

WacKEDmaN, 2021-01-27:
ok... yup looks like arduino had set compiler warnings to 'All'....
ive seen warnings before but never seen them handled as errors!
fyi im on arduino 1.18.13, ESP32 core (tried multiple) currently 1.0.5-rc6

fdivitto, 2021-01-27:
> ok... yup looks like arduino had set compiler warnings to 'All'....
> ive seen warnings before but never seen them handled as errors!

yes, it is the default in platform.txt:

```
compiler.warning_flags.more=-Wall -Werror=all
                                                             ^^^^^^^^^^^

```