fdivitto/FabGL #97: VNC client

niutech, 2021-02-23:
Is it possible to use FabGL Dev Board as a VNC thin client, using e.g. [arduinoVNC](https://github.com/Links2004/arduinoVNC) and WiFi? How to do it?

fdivitto, 2021-06-22:
I'm implementing that, starting from scratch (not from arduinoVNC). Initial tests showed me two major important limitations:
- display size
- number of colors

Reducing resolution we can have more colors and viceversa. Unfortunately modern operating systems use a lot of colors and high resolutions. I can assure you that at 640x480 x 16 colors the result is horrible!

I'll probably continue to implement it just for the fun to do it, but I don't know when it will be ready...:-)

niutech, 2021-06-22:
Thanks a lot for your effort! Does the problem lie in lack of memory? RFB supports various encodings, like RRE, TRLE and Hextile. Maybe using tiles would optimize the memory usage, updating only changed parts of the screen instead of the full framebuffer?
If I were to choose between size and colors, I would choose the former, because 1024x768x4 colors is still more usable than 320x240x32 colors.

fdivitto, 2021-06-22:
Encoding is useful transferring data, so it only affects speed: we need to maintain a frame buffer of what is displayed. Of course we update just the changed parts, but we still need to maintain a full screen buffer (what to send every scan line to the VGA out?). 1024x768x4 is allocatable but 4 colors are not enough.
Look at following images. First image is the original screen, second image is what we can see using ESP32 output, using 8 colors (not 4...):

[image]
[image]

Using 8 colors we can obtain up to 800x600, with the results just showed.
I've also tried color dithering: unfortunately it is good for pictures, but not for text and UI elements, so results are even worse.

niutech, 2021-06-22:
Dithering can help much with low color depth images, see your screenshot converted to dithered 4 colors:

[image]

and dithered grayscale:

[image]

For example, UltraVNC viewer has  options: `4greycolors` or even `2greycolors`. Some VNC servers also allow to remove wallpaper, which would further help.

It would be great if I could choose the XGA resolution even with monochrome (used with high contrast mode).

netpipe, 2021-08-12:
https://github.com/fdivitto/FabGL/issues/139

closed 2021-10-04 by fdivitto