fdivitto/FabGL #100: [Feature request] bold-attribut as different color and keyboard-click-sound

guidol70, 2021-03-09:
Hi,

as you wish -  here is/are my feature-request(s):

1.) It would be nice if you could add an option to replace the font-bold-attribut with a different color (like bold is shown as (non-bright) orange while the foreground text-color is green)
(see wordstar picture attachment)
This is an option in the puTTY-Terminal for Windows/Linux.
There is also the option to select the cursor-color ;)

2.) Another option I would like to see/hear would be the option for a key-click-sound like on the ATARI XL (same sound for every key) or a little bit better like on the Oric Atmos where the normal key have the samen sound, but the ENTER/RETURN-key does have another (deeper) sound ;)
[image]

fdivitto, 2021-03-14:
Please take a look at Terminal.setColorForAttribute() new method. Unfortunately just colors defined in Color enum are usable, so no Orange, unless you are using VGA16Controller and you change the palette with VGA16Controller.setPaletteItem().

fdivitto, 2021-03-14:
About sound on clik, with the last commit you may do this:

```c++
  Terminal.onVirtualKey = [&](VirtualKey * vk, bool keyDown) {
    if (keyDown)
      Terminal.soundGenerator()->playSound(SquareWaveformGenerator(), (*vk == VirtualKey::VK_RETURN ? 1000 : 2000), 4);;
  };
```

...or maybe this...

```c++
  Terminal.onVirtualKey = [&](VirtualKey * vk, bool keyDown) {
    if (keyDown)
      Terminal.soundGenerator()->playSound(VICNoiseGenerator(), (*vk == VirtualKey::VK_RETURN ? 90 : 120), 10);;
    Terminal.soundGenerator()->playSound(SquareWaveformGenerator(), (*vk == VirtualKey::VK_RETURN ? 1000 : 2000), 4);;
  };
```

I'm not sure if Oric Atmos click is a square wave or just noise... ;-)

guidol70, 2021-03-15:
@fdivitto **WOW** - the code worked in the 1st place when inserted in my RunCPM :)
I used your "**yellow-bright**"-example from the **SimpleTerminalOut.ino**
`Terminal.setColorForAttribute(CharStyle::Bold, Color::BrightYellow, false);`

and the "maybe this" example from above.

Compiled without errors in the first place ;) - for me a great experience now! 
**Many Thanks!!**

Now the longer part for me
- make it look tidy in "my" source and comment the lines for later aha-effects :)
- try to generate a valid configdialog for saving bold-color and enable/disable key-click
  and reloading/using this correct in the normal RunCPM-Application

closed 2021-03-15 by fdivitto

guidol70, 2021-10-08:
> VGA16Controller.setPaletteItem()
> Please take a look at Terminal.setColorForAttribute() new method. Unfortunately just 
> colors defined in Color enum are usable, so no Orange, unless you are using 
> VGA16Controller and you change the palette with VGA16Controller.setPaletteItem().

@fdivitto 
maybe you should change **for the example** in the docu for [SetPaletteItem](http://www.fabglib.org/classfabgl_1_1_v_g_a16_controller_a35a04bb9738fc5fc7f6e3fb4cbe90f37.html#a35a04bb9738fc5fc7f6e3fb4cbe90f37)
from 
displayController.setPaletteItem(0, RGB(255, 0, 0));
to
displayController.setPaletteItem(0, RGB**888**(255, 0, 0));

because on compile I did get the error **RGB is not defined** and it **worked with RGB888** like in the vga16controller.cpp

```
// Set Color 15 to AMBER-Orange for Retro-Display
// https://rgbcolorcode.com/color/amber
// RGB(255,191,0)
DisplayController.setPaletteItem(15, RGB888(255, 191, 0));
```

guidol70, 2021-11-08:
@fdivitto 
I try to set the AMBER-color in the confgialog.h of the ANSI-Terminal for the VGA16Controller (if I use a VGA16Controller resolution).

But the command isnt recognized, because the DisplayController is tied to BitmappedDisplayController also when DisplayController is redefined with
```
case ResolutionController::VGA16Controller:
DisplayController = new fabgl::VGA16Controller;
```

I dont understand why I cant do
```
case ResolutionController::VGA16Controller:
DisplayController = new fabgl::VGA16Controller;
DisplayController->setPaletteItem(9, RGB888(255, 191, 0));
break;
```

Do you got a idea/sourceline?

guidol70, 2021-11-10:
@fdivitto 
I did make me a terminal for my needs ;)
I took the Terminal-parts out of RunCPM VGA32 and some parts of the 
FabGL-ANSI-Terminal to get a 80x25 Terminal using the VGA16Controller.

There I could set a part of my palette to orange ;)

Named this little the fork **GLTerm** - if anyone want to test its get at
https://github.com/guidol70/RunCPM_VGA32/tree/main/GLTerm_VGA32