fdivitto/FabGL #99: PCEmulator does not compile

PQCraft, 2021-03-09:
```
In file included from /home/pqcraft/Arduino/libraries/FabGL/src/devdrivers/keyboard.h:37:0,
                 from /home/pqcraft/Arduino/libraries/FabGL/src/terminal.h:49,
                 from /home/pqcraft/Arduino/libraries/FabGL/src/fabgl.h:290,
                 from sketch/bios.h:26,
                 from sketch/bios.cpp:24:
/home/pqcraft/Arduino/libraries/FabGL/src/comdrivers/ps2device.h: In member function 'void BIOS::setKeyboardTypematicAndDelay()':
/home/pqcraft/Arduino/libraries/FabGL/src/comdrivers/ps2device.h:107:8: error: 'bool fabgl::PS2Device::sendCommand(uint8_t, uint8_t)' is protected
   bool sendCommand(uint8_t cmd, uint8_t expectedReply);
        ^
bios.cpp:516:44: error: within this context
     if (!m_keyboard->sendCommand(0xF3, 0xFA))
                                            ^
In file included from /home/pqcraft/Arduino/libraries/FabGL/src/devdrivers/keyboard.h:37:0,
                 from /home/pqcraft/Arduino/libraries/FabGL/src/terminal.h:49,
                 from /home/pqcraft/Arduino/libraries/FabGL/src/fabgl.h:290,
                 from sketch/bios.h:26,
                 from sketch/bios.cpp:24:
/home/pqcraft/Arduino/libraries/FabGL/src/comdrivers/ps2device.h:107:8: error: 'bool fabgl::PS2Device::sendCommand(uint8_t, uint8_t)' is protected
   bool sendCommand(uint8_t cmd, uint8_t expectedReply);
        ^
bios.cpp:519:67: error: within this context
     m_keyboard->sendCommand(i8086::BL() | (i8086::BH() << 5), 0xFA);
                                                                   ^
exit status 1
within this context
```

guidol70, 2021-03-09:
I think you only got the release v1.02 in your Arduino-FabGL-Library path, but you need to download the FabGL-master.zip at the github page and extract it to the FabGL-library path of your Arduino-IDE:
[HARDISK]/Users/[Username]/Documents/Arduino/libraries/FabGL/

I compiled it without problem (OK, without SDCard inserted) with Arduino IDE 1.8.13:
`Sketch uses 494673 bytes (37%) of program storage space. Maximum is 1310720 bytes.
Global variables use 20220 bytes (6%) of dynamic memory, leaving 307460 bytes for local variables. Maximum is 327680 bytes.
esptool.py v2.6
Serial port COM12
Connecting....
Chip is ESP32-PICO-D4 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, Embedded Flash, VRef calibration in efuse, Coding Scheme None
MAC: 50:02:91:a1:01:14
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 8192 bytes to 47...
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 2048.0 kbit/s)...
Hash of data verified.
Compressed 17392 bytes to 11186...
Wrote 17392 bytes (11186 compressed) at 0x00001000 in 0.2 seconds (effective 828.2 kbit/s)...
Hash of data verified.
Compressed 494784 bytes to 238057...
Wrote 494784 bytes (238057 compressed) at 0x00010000 in 4.6 seconds (effective 853.3 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 128...
Wrote 3072 bytes (128 compressed) at 0x00008000 in 0.0 seconds (effective 522.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
`

PQCraft, 2021-03-09:
Ah @guidol70 Thank you :)

closed 2021-03-09 by PQCraft