fdivitto/FabGL #102: [HINT] Timer issue for PCEmulator

guidol70, 2021-03-10:
for speed testing I did try to measure the time in GW-BASIC (or QBASIC) with the variables
TIMER and TIME$, but after the TIMER does reach 65.47 (seconds) the TIME$ starts/resets to 00:00:00 and also TIMER is reseting.

Normally TIMER only does count upwards and TIME$ has the actual time of is counting also upwards (slower than TIMER).

You could see the issue by running this short program:
10 PRINT TIMER, TIME$
20 IF TIME$="00:00:00" THEN END
30 GOTO 10

fdivitto, 2021-04-10:
I've just implemented the RTC clock and improved a bit the BIOS, so it should work now.
Thank you for tests!

closed 2021-04-10 by fdivitto

guidol70, 2021-04-11:
Its much better, but there is "space for optimization" because when I in GW-BASIC use
```
5 TIME$="13:25:00"
10 PRINT TIMER, TIME$
20 IF TIME$="00:00:00" THEN END
30 GOTO 10
```
then **TIME$** ouput starts at 12:59:58

and with 
`5 TIME$="12:07:00"`
then **TIME$** output starts at 11:59:58

So TIME$ is missing to aquire the minutes and seconds :(

The variable TIMER seems to be OK :) because the number is right for
PRINT (HOURS*3600)+(MIN*60)+SEC
like 46803 for 13:00:03

TIME$
https://hwiegman.home.xs4all.nl/gw-man/TIMES.html

TIMER
https://hwiegman.home.xs4all.nl/gw-man/TIMER.html

fdivitto, 2021-04-11:
Thanks, I forgot to save DX in INT 1A, subfunction 1. 
I never thought this project would improve my skills with 8086 assembler and writing a PC BIOS ;-)

guidol70, 2021-04-11:
Cool :) That did the job for my "problem". If you want you can keep this issue closed ;)