fdivitto/FabGL #103: SPIFFS with FabGL lbrary

phreda4, 2021-03-14:
I have problem with SPIFFS. If I list a directory before activating the library, it works, but if I try to list it afterwards a crash occurs.
Is there a way to temporarily disable vga signal generation (like zx81 and fast mode) get the files and restart the genaration?

fdivitto, 2021-03-14:
This is weird, some fabgl demos uses extensively SPIFFS without crashes. Please post some code to test.

phreda4, 2021-03-14:
yes, this is setup()

```
void setup()
{
Serial.begin(115200);Serial.write("ESP32 Forth Computer\n");
if (!SPIFFS.begin(true)) {
    Serial.println("An Error has occurred while mounting SPIFFS");
    return;
  }
/*------------this work OK
  File root = SPIFFS.open("/");
  File file = root.openNextFile();
  while(file){
      Serial.print("FILE: "); Serial.println(file.name());
      file = root.openNextFile();
  }
  file.close();
  root.close();
----------------*/  

mainTaskHandle = xTaskGetCurrentTaskHandle();
PS2Controller.begin(PS2Preset::KeyboardPort0); //PS2Preset::KeyboardPort0_MousePort1);

auto keyboard = PS2Controller.keyboard();
keyboard->setLayout(&fabgl::SpanishLayout);
        
DisplayController.begin();
DisplayController.setDrawScanlineCallback(drawScanline);

DisplayController.setResolution(QVGA_320x240_60Hz); // 40x30

r3init();
ccls(); 
cink(60);cprint("ESP32 ");cink(3);cprint("Forth ");cink(12);cprint("Computer ");ccr();

/* // -----------ERROR when uncoment
  File root = SPIFFS.open("/");
  File file = root.openNextFile();
  while(file){
      Serial.print("FILE: "); Serial.println(file.name());
      file = root.openNextFile();
  }
  file.close();
  root.close();
  ----------------------*/
}

```

fdivitto, 2021-03-15:
please can you show also "drawScanline" function?

phreda4, 2021-03-15:
yes

```
// pixel order 2301
void IRAM_ATTR drawScanline(void *arg,uint8_t *dest,int scanLine) {
byte c[2],bf,*a=dest;
uint32_t addrc=scanLine>>3;addrc=(addrc<<3)+(addrc<<5); //*40
uint32_t cc,*ac=&screen[addrc];
for (byte x=SCREEN_W;x!=0;x--)	{ // 80 40
  cc=*ac++;
  c[0]=((cc>>16)&0x3f)|DisplayController.m_HVSync; // I modify the lib for public 
  c[1]=((cc>>24)&0x3f)|DisplayController.m_HVSync;
  if (cc&0x80000000) {  // graphics x2
    bf=cc>>(((scanLine>>1)&3)<<2);  
    *a++=*a++=c[(bf>>1)&1];
    *a++=*a++=c[(bf>>0)&1];
    *a++=*a++=c[(bf>>3)&1];
    *a++=*a++=c[(bf>>2)&1];
    continue; }
  bf=romfont[((cc&0xff)<<3)+(scanLine&7)];    
  if (cc&0x40000000) { bf=~bf; }      // neg
  if (cc&0x800000) { bf=blink^bf; } // blink
  if (cc&0x400000) { if ((scanLine&7)==7) { bf=~bf; } } // underline?
  *a++=c[(bf>>5)&1];  
  *a++=c[(bf>>4)&1];  
  *a++=c[(bf>>7)&1];  
  *a++=c[(bf>>6)&1];
  *a++=c[(bf>>1)&1];      
  *a++=c[bf&1];  
  *a++=c[(bf>>3)&1];  
  *a++=c[(bf>>2)&1];
  }
if (scanLine==DisplayController.getScreenHeight()-1) { // signal end of screen
  vTaskNotifyGiveFromISR(mainTaskHandle, NULL); 
  if (millis()&0x100) { blink=0xff; } else { blink=0; } 
  }
}
```

I do a modification for access to sync bits. I have a screen of 40x30 32bits characters

every cell:

two color of 6bits, fore and back, 4 bits flag for type (graphics, invert,blink, underline)-complete the 16 upper bits, then, if is graphics a 4x4 pixel description or 8 lower bits character

the complete code is in https://github.com/phreda4/esp32-forth-computer

fdivitto, 2021-03-15:
code and data inside drawScanline should be in RAM (not PSRAM, not Flash), because it is called from inside an interrupt declared as IRAM_ATTR. Is "romfont" in Flash/PSRAM or RAM?
Other checks should be done for millis() and screen[].

phreda4, 2021-03-15:
all the data is declare in code, no malloc use, I think is in RAM.
I remove milis() and put outside and is the same, not work
I try to read more code, or try to add a not fresh vga state.
thank's a lot

fdivitto, 2021-03-15:
how is it declared romfont[]?

fdivitto, 2021-03-15:
romfont is stored in Flash because of "const":

"static const byte romfont[256*8]...."

phreda4, 2021-03-15:
ok.. thanks.. I don't know this

fdivitto, 2021-03-15:
actually you could even access Flash or PSRAM from an ISR declared with IRAM, but this requires you don't use temporarily SPIFFS (which disables Flash and PSRAM...).

closed 2021-03-15 by fdivitto