fdivitto/FabGL #104: [Question] usage of "Terminal.setColorForAttribute" in loadConfiguration

guidol70, 2021-03-15:
I successfully impletented Keyclick & configdialog for Keyclick, but for the 
Terminal.setColorForAttribute I cant seem to find the right syntax to use it
in loadConfiguration :(

I did try
Terminal.setColorForAttribute(CharStyle::Bold, Color::getBDColor(), true);
but getBDColor() isnt a member of Color:: - I think because getBDColor()
does result in a number and not the Text-name-color (which should be a member of
Color::

Have I to create a array with the names of is there any actual useable syntax already there?

My actual configdialog.h as attachment ;)
[RunCPM_configdialo_h.txt](https://github.com/fdivitto/FabGL/files/6142199/RunCPM_configdialo_h.txt)

fdivitto, 2021-03-15:
In your code getBDColor() is a static method of ConfDialogApp and not of Color, so you should write:

Terminal.setColorForAttribute(CharStyle::Bold, getBDColor(), true);

guidol70, 2021-03-15:
@fdivitto Thanks again! works like a charm - so my configdialog now can save keyclick&bold color and RunCPM does use it.

At first when I had the click working then no F12 nor BREAK (Reboot without pref-clear) did work, but after changing the Terminal.onVirtualKey/if-"loop" all keybombinations did work again.

PS: then it should also be right to use 
`Terminal.setColorForAttribute(CharStyle::Bold, ConfDialogApp::getBDColor(), true);`
in the .ino-firsttab?

fdivitto, 2021-03-15:
> PS: then it should also be right to use
> `Terminal.setColorForAttribute(CharStyle::Bold, ConfDialogApp::getBDColor(), true);`
> in the .ino-firsttab?

yes, of course after both "Terminal" and "preferences" have been initialized.

closed 2021-03-15 by fdivitto