fdivitto/FabGL #105: [Bug?] System does hang while using KeyClick with BREAK (Ctrl+PAUSE)

guidol70, 2021-03-15:
Before I included the KeyClick in RunCPM with FabGL my Reset-Routine via **VK_BREAK** did work.
Now with working KeyClick all normal Keys do work (also F12, ESC **and Ctrl alone**) but
with VK_BREAK (Ctrl+PAUSE) the Input does hang while using KeyClick :(

Found a workaround while debugging (wrote a [.] for every keypress in the KeyClick-Routine)

When I wrote with "Terminal.write" that debugging-stuff ALL keys did work - also VK_BREAK for reset *strange* :)
So the final workaround by now is to write a empty-String to the Terminal before the KeyClick like
**Terminal.write("");**

Dont know its my weird source or its a "bug"?
I do attach my .ino for reference ;)
[RunCPM_VGA32_v5_1_ino.txt](https://github.com/fdivitto/FabGL/files/6144465/RunCPM_VGA32_v5_1_ino.txt)

fdivitto, 2021-03-15:
do you mean that after you pressed CTRL+PAUSE, ESP doesn't reset and you get no more keys from keyboard?

fdivitto, 2021-03-15:
just one tip: if you don't handle a virtual key don't set "*vk = VirtualKey::VK_NONE;". So it should be inside the "if (!keyDown)", or just remove it because you are going to reset the ESP:

```c++
    if (*vk == VirtualKey::VK_BREAK) {
      if (!keyDown) {
        // releasing BREAK key to reboot
        Terminal.write("\r\n");
        Terminal.write("\r\n______________________________");
        Terminal.write("\r\nReboot requested via BREAK-Key");
        Terminal.write("\r\nRebooting in 6 seconds...");
        Terminal.write("\r\n______________________________");
        Terminal.write("\r\nCountDown: ");
        Terminal.write("[5] ");
        delay(1000);
        Terminal.write("[4] ");
        delay(1000);
        Terminal.write("[3] ");
        delay(1000);
        Terminal.write("[2] ");
        delay(1000);
        Terminal.write("[1] ");
        delay(1000);
        Terminal.write("[REBOOT]");
        delay(1000);
        ESP.restart();
      } 
      //*vk = VirtualKey::VK_NONE;    <== removed
    }
```

guidol70, 2021-03-16:
@fdivitto Yes - when the hangup occurs then the keyboard has no function anymore
and also a reconnect of the keyboard doesnt repair that state (but display is OK and Cursor is blinking - so not completly sytem-lock.
But because it was sometimes working and sometimes **not (also with my "workaround")** I came to the idea - yesterday when laying in the bed - that I have to check my other keyboards.
The problems above I had with a "CHERRY ML-4100 DE"

I have 4 PS/2-Keyboards (listed here **from the oldest to the newest**):
- CHERRY ML-4100 DE-LayOut 5V 10mA
- MS-TECH LT-300 DE-LayOut 5V 170mA (from around 2003)
- KYE Systems Corp - Genius GK-04006 UK-LayOut
- frisby FK-88QP TR-LayOut 5V 300mA (0.3A)

When I do a keyboard-stress-test - like a little child - and typing **"wild"** and **fast** on the keyboard the CHERRY locks up after around 5-20 keys
The MS-TECH isnt much better.
With the KYE Genius I could archive sometimes 40-120 Chars before lockup.
BUT the frisby-Keyboard (did buy it here in turkey 2-3 years ago) can type "wild" like 3x3 Commandlines (CP/M A: Prompt does switch after 3 Lines) and then it **may** only stuck/lockup if I do not made a small break. 
With a small break its mostly impossible to hang-/lockup the input.

I dont know if its a timing problem of the Keyboard-controller-chip or somthing like "roll-over" which is - I think - present at PS/2 Keyboard when pressing serveral keys at the same time.

With the frisby-Keyboard (and also KYE Genius) I hadnt the problem with the BREAK-Key.

So also my source doenst really seem to be the problem. But I dont know why ther compatibility for the keyboards is so different.

Is there any debug option for the PS/2-Keyboard routines that produce output which I could test to improve the handling of the keyboard-queue?

BTW: I also removed the VK_NONE and included your last commits for "removed suspendVirtualKeyGeneration()" in my last compile. So that was also not the fault ;)

guidol70, 2021-03-16:
Found a 5th keyboard - my trusty old IBM Model M (55-07113261F) from 1992, but this also lockup after 20-40 chars while "wild" typing :(

fdivitto, 2021-03-16:
PS/2 controller has been rewritten from scratch since 11th March. I asked help for tests (https://github.com/fdivitto/FabGL/issues/86) but nobody replied :-(
So this may be related to the new PS/2 controller implementation.

To find the problem we should test a minimal application, for example "Others/KeyboardStudio". Please could you repeat such tests on this application?

fdivitto, 2021-03-16:
please wait, I'll tell you when I'm ready you perform the tests...

fdivitto, 2021-03-17:
please could you test last commit?

guidol70, 2021-03-17:
@fdivitto Heureka! - your last commit "PS2Controller: workaround for interrupt stall"
**_SOLVED_** the problem with my older keyboard! - **YOU ARE GREAT!** ;)

fdivitto, 2021-03-17:
well, thank you! Do you confirm that now following problems are fixed?
- CTRL+BREAK stops keyboard inputs
- no input from keyboard after "intensive" keypresses...

guidol70, 2021-03-17:
From my side CONFIRMED that the problems are fixed ;)

closed 2021-03-17 by fdivitto