fdivitto/FabGL #107: Nano-X / Microwindows port for FabGL

niutech, 2021-03-17:
Is it possible to port [Nano-X Window System](https://github.com/ghaerr/microwindows) (f.k.a. Microwindows) for your fantastic FabGL library, so that it would be possible to run existing Nano-X, X11 and FLTK apps on ESP32? What would it take to do it?

Thanks!

fdivitto, 2021-03-17:
I really don't know. I don't know Nano-X/Microwindows memory requirements, speed, etc... For sure you can create mouse, keyboard and video drivers for Nano-X/Microwindows with fabgl.

niutech, 2021-03-17:
From the [FAQ](http://www.microwindows.org/faq.html):
>The graphics engine is capable of running on any system that supports readpixel, writepixel, drawhorzline and drawvertline, and setpalette.  Blitting support is optional, but if implemented allows enhanced functionality.  All bitmap, font, cursor and color support is implemented on top of these routines.  Support for 8, 15, 16, 24 and 32 bit truecolor systems as well as 1, 2, 4 and 8bpp palletized systems is implemented.
>The Nano-X Window System is extremely portable, and completely written in C, although some routines have been recoded in assembly for speed.  It has been ported to the Intel 16 and 32 bit cpu's, as well as MIPS R4000 (NEC Vr41xx) StrongARM and PowerPC chips found on popular handheld and pocket PC's.  There have been rumors of an SH3 port as well.
>On 16 bit systems, the entire system, including screen, mouse and keyboard drivers runs in less than 64k. On 32-bit systems, support includes proportional fonts and applications are typically less than 100k.

Does ESP32 meet these criteria?

fdivitto, 2021-03-17:
> The graphics engine is capable of running on any system that supports readpixel, writepixel, drawhorzline and drawvertline, and setpalette.  Blitting support is optional, but if implemented allows [...]

FabGL meets above criteria.
Anyway specific tests should be done on following part:

> The Nano-X Window System is extremely portable, and completely written in C, although some routines have been recoded in assembly for speed.  It has been ported to the Intel 16 and 32 bit cpu's, as [...]

I hope someone has the ability and the time to do it.

closed 2021-03-31 by fdivitto