fdivitto/FabGL #108: [Bug?] Reset ANSI-Terminal Alt+Ctrl+F12 isnt working (for me)

guidol70, 2021-03-22:
I did try to fork the AnsiTerminal.ino and edit it but did get a 500 Error from github, so I will make open a issue, because with the actual .ino code of the AnsiTerminal the Clear-Reset isnt working for me :(

I ended up with editing the part for Alt+Ctrl+F12 and it start to work for me with the following code-part:
`      } else {
        // pressing CTRL + ALT + F12, reset parameters and reboot
        if ((Terminal.keyboard()->isVKDown(VirtualKey::VK_LCTRL) || Terminal.keyboard()->isVKDown(VirtualKey::VK_RCTRL)) &&
            (Terminal.keyboard()->isVKDown(VirtualKey::VK_LALT) || Terminal.keyboard()->isVKDown(VirtualKey::VK_RALT)) &&
            (Terminal.keyboard()->isVKDown(VirtualKey::VK_F12))) {
          Terminal.deactivate();
          preferences.clear();
          // show reboot dialog
          auto rebootApp = new RebootDialogApp;
          rebootApp->run(DisplayController);
        }
      }`

See the additional line for VK_F12

This works for me on a normal-size-keyboard where F12 is a seperate key,
but didnt work an a small PS/2-keyboard where I have to press Alt+Ctrl+[Fn]+F2[F12]

Did the clear/restart work with your old code and your keyboard the last times?

Kind regards
Guido

fdivitto, 2021-03-22:
It is strange, because VK_F12 is already checked before entering in the nested IF, so I guess the bug comes from something other.

I've just posted a commit where a new delegate has been added, named "onVirtualKeyItem". It differs from onVirtualKey because now the SHIFTs status (CTRL, ALT, SHIFT, etc...) are embedded into the delegate parameters. The problem of isVKDown() method is that it returns "current" key status, not the status at the time the key was actually down.

Unfortunately I haven't a keyboard without F12, so I cannot test.

guidol70, 2021-03-22:
Have here a short debug out of the Keyboard-Studio, but the KeyboardStudio doesnt seem to recognize the [FN]-Key which should switch between F2 and F12 :(
[Keyboard_Debug.txt](https://github.com/fdivitto/FabGL/files/6185076/Keyboard_Debug.txt)

fdivitto, 2021-03-22:
Please could you do this test for me?
Upload this sketch (this is the same of KeyboardScancodes example) and just press following combinations:
- FN + F2
- CTRL + FN + F2
- CTRL + ALT + FN + F2

Of course show me your results.

```c++
#include "fabgl.h"

fabgl::VGATextController DisplayController;
fabgl::Terminal          Terminal;
fabgl::PS2Controller     PS2Controller;

void xprintf(const char * format, ...)
{
  va_list ap;
  va_start(ap, format);
  int size = vsnprintf(nullptr, 0, format, ap) + 1;
  if (size > 0) {
    va_end(ap);
    va_start(ap, format);
    char buf[size + 1];
    vsnprintf(buf, size, format, ap);
    Serial.write(buf);
    Terminal.write(buf);
  }
  va_end(ap);
}

void printHelp()
{
  xprintf("\e[93m\n\nPS/2 Keyboard Scancodes\r\n");
  xprintf("Chip Revision: %d   Chip Frequency: %d MHz\r\n", ESP.getChipRevision(), ESP.getCpuFreqMHz());

  printInfo();

  xprintf("Commands:\r\n");
  xprintf("  q = Scancode set 1  w = Scancode set 2\r\n");
  xprintf("  l = Test LEDs       r = Reset keyboard\r\n");
  xprintf("Various:\r\n");
  xprintf("  h = Print This help\r\n\n");
  xprintf("Use Serial Monitor to issue commands\r\n\n");
}

void printInfo()
{
  auto keyboard = PS2Controller.keyboard();

  if (keyboard->isKeyboardAvailable()) {
    xprintf("Device Id = ");
    switch (keyboard->identify()) {
      case PS2DeviceType::OldATKeyboard:
        xprintf("\"Old AT Keyboard\"");
        break;
      case PS2DeviceType::MouseStandard:
        xprintf("\"Standard Mouse\"");
        break;
      case PS2DeviceType::MouseWithScrollWheel:
        xprintf("\"Mouse with scroll wheel\"");
        break;
      case PS2DeviceType::Mouse5Buttons:
        xprintf("\"5 Buttons Mouse\"");
        break;
      case PS2DeviceType::MF2KeyboardWithTranslation:
        xprintf("\"MF2 Keyboard with translation\"");
        break;
      case PS2DeviceType::M2Keyboard:
        xprintf("\"MF2 keyboard\"");
        break;
      default:
        xprintf("\"Unknown\"");
        break;
    }
    xprintf("\r\n", keyboard->getLayout()->name);
  } else
    xprintf("Keyboard Error!\r\n");
}

void setup()
{
  Serial.begin(115200);
  delay(500);  // avoid garbage into the UART
  Serial.write("\r\n\nReset\r\n");

  DisplayController.begin();
  DisplayController.setResolution();

  Terminal.begin(&DisplayController);
  Terminal.enableCursor(true);

  PS2Controller.begin(PS2Preset::KeyboardPort0, KbdMode::NoVirtualKeys);

  printHelp();
}

void loop()
{
  static int clen = 1;
  auto keyboard = PS2Controller.keyboard();

  if (Serial.available() > 0) {
    char c = Serial.read();
    switch (c) {
      case 'h':
        printHelp();
        break;
      case 'r':
        keyboard->reset();
        printInfo();
        break;
      case 'l':
        for (int i = 0; i < 8; ++i) {
          keyboard->setLEDs(i & 1, i & 2, i & 4);
          delay(1000);
        }
        delay(2000);
        if (keyboard->setLEDs(0, 0, 0))
          xprintf("OK\r\n");
        break;
      case 'q':
        keyboard->setScancodeSet(1);
        xprintf("Scancode Set = %d\r\n", keyboard->scancodeSet());
        break;
      case 'w':
        keyboard->setScancodeSet(2);
        xprintf("Scancode Set = %d\r\n", keyboard->scancodeSet());
        break;
    }
  }

  if (keyboard->scancodeAvailable()) {
    int scode = keyboard->getNextScancode();
    xprintf("%02X ", scode);
    if (scode == 0xF0 || scode == 0xE0) ++clen;
    --clen;
    if (clen == 0) {
      clen = 1;
      xprintf("\r\n");
    }
  }

}
```

guidol70, 2021-03-23:
Here the test results of the compact keyboard (Cherry G84-ML4100)
For me it looks like it doenst see FN when in combination with Ctrl/Alt - so it gives a F2 only
[Keyboard_Scantest_2303202.txt](https://github.com/fdivitto/FabGL/files/6187537/Keyboard_Scantest_2303202.txt)
[image]

closed 2021-03-27 by fdivitto