fdivitto/FabGL #110: [Enhancement?] using USB-serial Port for ANSI-Terminal - code already attached :)

guidol70, 2021-03-25:
Since you Debug-Output at the "old" KeyboardStudio I was inspired using the USB-serial port.
First I did create a like mirror Terminal to USB-serial output RunCPM version where selectable the input can created on PS/2-keyboard or over serial-session over USB-serial.
Does work fine and looks great :)

Now I was thinking about this USB-serial option and edited your ANSI-Terminal also for switchable normal-serial/USB-serial usage.

I created a Select-Box in the Configpage via the configdialog.h

The TTGO VGA32 (which I use) appears as /dev/ttyUSB0 on my linux and I could tie it to /dev/ttyUSB1 with
socat  /dev/ttyUSB0,raw,echo=0,crnl,b115200 /dev/ttyUSB1,raw,echo=0,crnl,b115200
for using the ANSI-Terminal with my Z80-MBC2

The other idea would be to get a Linux-Console at the ANSI-Terminal with
/sbin/getty -L 115200 ttyUSB0 ansi

This kinda works, but it seems to have not the full control (maybe I should another Terminal-Emulation?)

How do you think about this option as enhancement?

I will attach the .ino for USB-Serial-ANSI-Terminal here as .ZIP 
[AnsiTerminal_USBSer.zip](https://github.com/fdivitto/FabGL/files/6203246/AnsiTerminal_USBSer.zip)

guidol70, 2021-03-26:
to get around the Linix-console control-problems **DONT USE**
/sbin/getty -L 115200 ttyUSB0 ansi

**USE INSTEAD**

**enable at boot:**
systemctl enable serial-getty@ttyUSB0.service

**disable:**
systemctl disable serial-getty@ttyUSB0.service

**enable temporary:**
sudo service serial-getty@ttyUSB0 start

**enable the temporary:**
sudo service serial-getty@ttyUSB0 stop

fdivitto, 2021-04-01:
thank you, just implemented!

closed 2021-04-01 by fdivitto

guidol70, 2021-09-17:
as a new Feature (**USBSerialFilter**) I implemented the option to start/stop the simultaneously output to VGA-Terminal/USBSerial, so that the USBSerial-Output ca be used to control a external device and you havent to filter the complete display-output ;)

This is implemented in **Rev GL20210917.0** of RunCPM VGA32 available under
https://github.com/guidol70/RunCPM_VGA32/tree/main/v5_6