fdivitto/FabGL #112: Trying to compile latest RunCPM

hjkit, 2021-04-09:
I am getting the following error messages using the latest FabGL

Arduino: 1.8.13 (Linux), TD: 1.53, Board: "TTGO T7 V1.4 Mini32, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

./Arduino/libraries/FabGL/src/comdrivers/ps2controller.cpp: In function 'void fabgl::replace_placeholders(uint32_t, int, bool, gpio_num_t, gpio_num_t, bool, gpio_num_t, gpio_num_t)':
./Arduino/libraries/FabGL/src/comdrivers/ps2controller.cpp:1005:54: error: 'rtc_io_number_get' was not declared in this scope
     int port0_clkIO = rtc_io_number_get(port0_clkGPIO);

./Arduino/libraries/FabGL/src/comdrivers/ps2controller.cpp:1007:28: error: 'rtc_io_desc' was not declared in this scope
     CLK_rtc_gpio_reg[0]  = rtc_io_desc[port0_clkIO].reg;
                            ^
./Arduino/libraries/FabGL/src/comdrivers/ps2controller.cpp:1025:54: error: 'rtc_io_number_get' was not declared in this scope
     int port1_clkIO = rtc_io_number_get(port1_clkGPIO);
                                                      ^
./Arduino/libraries/FabGL/src/comdrivers/ps2controller.cpp:1027:28: error: 'rtc_io_desc' was not declared in this scope
     CLK_rtc_gpio_reg[1]  = rtc_io_desc[port1_clkIO].reg;
                            ^
Any hints, tips

fdivitto, 2021-04-10:
This is a problem of current release of FabGL embedded in Arduino. The version you can download we should fix it.

Soon will be released a new release available from inside the Arduino IDE.

guidol70, 2021-04-10:
@fdivitto @hjkit 
Hi Fabrizio ;)
for hjkit I also tested the RunCPM compile with the actual FabGL-master.zip in the Arduino-IDE and did get **WITH LINUX** the same error (see attached file) for the ps2controller.cpp
but I didnt get that error when compiling the same .ino with Windows 10 and the Arduino-IDE (both 1.8.13)
I tested with the FabGL-master.zip from the 01.04.2021 / 06.04.2021 and the 10.04.2021
Allways the same - the error is only iwth Linux :( and you know I did compile RunCPM with FabGL very often :)
I have the same SdFAT library and the same download URL for the ESP32-Board-support.

Seems that the compiler under Linux mean the scope isnt the same as under Windows :(
Do you got an idea - beside the full new release to the Arduino-IDE?

I doesnt know a error which should be only under Linux.

[Linux_Error_compile.txt](https://github.com/fdivitto/FabGL/files/6290402/Linux_Error_compile.txt)

fdivitto, 2021-04-10:
Thank you, I'll check it.

guidol70, 2021-04-10:
@fdivitto @hjkit 
I did found the problem at frist sight when doing a compile in "**Verbose-Mode**" on Windows 10 against the newly installed Linux Arduino-IDE :)

The "**problem**" is that the new Linux-Install bring us **ESP32 HW Support v1.0.6** and Fabrizio did recently (on **26. March 2021**) **"only" v1.0.5**:
https://github.com/fdivitto/FabGL/commit/e9242411b2385c6faed2697a5e500f759f1c2506

So after removing v1.0.6 via the Arduno IDE Board Manager and then installing v1.0.5
the compile did work better....BUT because it was a new Installation for me there were some little quirks to do:
I had "**python missing in the PATH**" - that means python3 was install , but it was missing:
`sudo ln -s /usr/bin/python3 /usr/bin/python`

then python was missing the installation of Pyserial - so I did:
`apt install python3-serial`

then Pyserial was missing the access-right for /dev/ttyUSB0 :) - so I did:
`sudo usermod -a -G dialout guido`
(you have to do that for your username  - which do start the Arduino-IDE)

after that you have to logout/login for activating the right or reboot :)

After these steps the .ino did compile fine and the tranfer to the TTGO VGA43 was successful!

I hope this will also resolve your problems ;)

fdivitto, 2021-04-10:
Last commit should fix compatibility with Ardunio Library 1.0.6

guidol70, 2021-04-10:
@fdivitto @hjkit 
I tested with ESP32 Board-Library v1.0.6 and it did compile RunCPM without issues - Thanks! :)
I think hjkit could do the Linux test for v1.0.6 ;)

hjkit, 2021-04-11:
Thanks for all the effort. I am really thankful. Now I can compile the latest version without any error but no input will be accepted :-(

Version RunCPM_v5_1-master_TTGO_VGA32_05032021 ist working, all later ones don't accept any Keyboard input :-(

fdivitto, 2021-04-11:
> Thanks for all the effort. I am really thankful. Now I can compile the latest version without any error but no input will be accepted :-(

I cannot replicate on my side. Could you test other examples (like LoopbackTerminal or Others/KeyboardVirtualKeys)?

hjkit, 2021-04-11:
I tried the following test
HardwareTest ok  keyboard input is working
LoopbackTerminal   keyboard error
SimpleTerminalOut ok
AnsiTerminal not working
GraphicalUserInterface ok
Altair8800 no keyboard input
What else should I test ?

hjkit, 2021-04-11:
Could it be it has something to to with a German keyboard?
I tried again Altair8800 with FabGL 1.0.2. Everything seemed fine. Then I called the options menu with the function key and switched to German keyboard. Since then no input work is accepted. F12 is still recognized, but I can no longer select any options (by ASCII Keys). Even a renewed download did not change anything.

guidol70, 2021-04-11:
@hjkit 
Could you see if "USB Serial control" is enabled on the Boot-Screen?
Then the normal CP/M prompt is controlled via the USB-serial port at 115.200Baud.

But for the first test open with F12 the config menu and mobe with the Tab-Key from selection to selection until USBSerCtl and deactive it with pressing the Spacebar.

If it then not will automatically reboot after a F10 save-config then reboot manually.
Then the PS/2-Keyboard should have control over the CP/M-prompt.

German is only the keyboard-layout, so should be no problem if you got something like a UK-Keyboard connected.

Could you move&select something in the Config-Menu?

RunCPM_v5_1-master_TTGO_VGA32_05032021 is very old and @fdivitto did wrote some new routines for the keyboard, but I have 5 PS/2 keyboard from differnt brands and all do work for input.
You do use the violet-keyboard-connector on the VGA32 and not the green for the mouse?
Do you got another PS/2-keyboard?

If USBSerCtl is enabled at the boot-screen its only for the configuration....
Via USBSerCtl you dont have a KeyClick, Reboot-Key and no chance to open ConfigMenu.

Does the Reboot-Key BREAK (=Ctrl+PAUSE) work? PAUSE is right beside the Scroll-Lock-Key

Which PS/2 Keyboard do you have? Which keyboard-Layout ?
Is there a Info about how much mAh the voltage draw on the underside?
Did you try another power-supply? Or how do you power the VGA32?
If from the PC try to use another USB-Port.

hjkit, 2021-04-12:
@guidol70

To answer some off your questions

As RunCPM_v5_1-master_TTGO_VGA32_05032021 ist should be a sufficient indicator that mouse and keyboard are connected to the right input-ports.
I am not color blind and I can still distinguish between green and violet (smiley) 
I have tried different usb prots and cables.
power is supplied via USB cable
I have not tried another power supply because Version RunCPM_v5_1-master_TTGO_VGA32_05032021 working
I have tried  ESP32 Dev Modul and TTGO T7 V1.4 Mini32
I am using IBM PC PS2 Keyboard which works for 3 other PCs
Keyboardscancode show
PS/2 Keyboard
Chip Revision: 1 Chip Frequency: 240 Mhz
Device ID = MF2 keyboard"
I don't have another PS Keyboard

I have tried it once more this morning focusing specially on your hint
regarding USBSerCtl

USBSerCtl was enabled
Reboot-Key BREAK and F12 are working
I disabled the USBSerCtl
still no character input from keyboard

After a reboot
Reboot-Key BREAK and F12 are no longer working

I give up and will stay with Version RunCPM_v5_1-master_TTGO_VGA32_05032021 or order another TTGO

P.S.
Can you send my you working binaries so that I try them ?

guidol70, 2021-04-12:
@hjkit 
> P.S.
> Can you send my you working binaries so that I try them ?

[RUNCPM_VGA32__binary_09042012.zip](https://github.com/fdivitto/FabGL/files/6297499/RUNCPM_VGA32__binary_09042012.zip)

hjkit, 2021-04-13:
@guidol70
With a different keyboard and Guidos binaries, I've come a step further. It nearly always works now. Sometimes "no keyboard" appears in the boot info, and when I do a power recycle, it is working again.
I also applied an external voltage (3.3V) via the GPIO pin strip, then the board does not boot at all.  I use a KVM switch for the keyboard and the mouse for my 3 PCs to share.
With the KVM switch, the keyboard is almost never recognized with RunCPM, although all tests programs (KeyboardScanCode, KeyboardVirtualKeys, GraphicalUserInterface) do work with the KVM keyboard.
If I boot with the external keyboard and change to the KVM keyboard after the board is up it almost always works.
I think something in the keyboard recognition is causing problems when I am using my KVM keyboard.
I have ordered 2 more TTGO and will then test if something will change.

Thanks again for the your support.

fdivitto, 2021-04-13:
> I have tried ESP32 Dev Modul and TTGO T7 V1.4 Mini32

I suppose you haven't tried a development board or TTGO VGA32, right?

closed 2021-06-22 by fdivitto