fdivitto/FabGL #113: [Question] Support for the RST and IO36 button on the TTGO VGA32 v.14

guidol70, 2021-04-13:
I recognized that your PC-Emulator does work (will do a Reset) if the RST-Button under the Keyboard-connector is pressed and I did see/read that you are using GPIO12 for preferences.clear with the ANSI-Terminal sketch.

Also I did understand/read that GPIO12 shouldnt be used when the MicroSD-Card is in use (or is this only GPIO02?) because of the boot-init(flash-voltage.

So my Questions:
Is the RST-Button GPIO12? Or how (when you didit program) did you get the RST-Button working for the PC-Emulator?

If RST is GPIO12 I couldnt/shlouldnt use it. Then I would support the secind button (GP)IO36  under the Mouse.Connector.

But my follwing (shorted) code didnt work :( It seems that the sketch (based on a older ANSI-Terminal-sketch) never execute/reach the "void loop":

```
#define RESET_PIN 36

void setup(void)	{
   pinMode(RESET_PIN, INPUT);
					}

void loop(void) 	{
  if (digitalRead(RESET_PIN) == 1)
     _reboot;
					}
```

I have "successful" tested (GP)IO36 in the "void setup" section - that means I had to press IO36 to complete the execution of the "void setup" section.

Could you please give me a hint where I can put such a reset-routine?

PS: _reboot is my own defined reboot function (some display/sound/countdown) ;)

guidol70, 2021-04-13:
I found out (in "void setup") that when (GP)IO36 is pressed it isnt HIGH, but LOW:
```
if (digitalRead(RESET_PIN) == HIGH) {
   _puts("GPIO36 HIGH detected.\r\n");
                                    }
                                    
if (digitalRead(RESET_PIN) == LOW)  {
   _puts("GPIO36 LOW detected.\r\n");
                                    }
```
but the following doesnt work in "void loop":
`if (digitalRead(RESET_PIN) == LOW) { _reboot; }`

guidol70, 2021-04-13:
So - I sorted the .ino source of the frist tab and his brackets for better reading ;)

Now I know why I didnt read "void loop" :)
The .ino never reaches "void loop" because the execution is intercepted inside the "void setup" after the following instructions:
- Loading CCP
- Output of CCP-Head
- Patching CP/M

So the control is handed over to the CCP - so no chance for my (GP)IO36 :(

fdivitto, 2021-04-13:
on TTGO VGA32 v1.4, RTS button is EN pin (ESP32 reset), while the other button is IO36.
IO12 and IO2 are used for the SD Card (other than IO13 and IO14).

>>>Could you please give me a hint where I can put such a reset-routine?

why not a GPIO interrupt?

guidol70, 2021-04-13:
> why not a GPIO interrupt?

because I didnt know such things - Iam not a real Arduino-code-programmer ;)
I googled the "GPIO Interrupt" and got it working after a few minutes with:

```
void _esp_reboot(void) {
        _clrscr();
        ESP.restart();
                                      }

attachInterrupt(digitalPinToInterrupt(RESET_PIN), _esp_reboot, FALLING);
```

Thanks!

closed 2021-10-04 by fdivitto