fdivitto/FabGL #115: PS2 Keyboard not working for some examples

magore, 2021-05-09:
Keyboard input does not work for several examples.
I am using Arduino 1.8.14 and esp32 1.0.6 with a vga32_v1.4 board.
Tested on May 9th with fresh git clone of you repo: d43d3e93fd2815c1253e4c49790e1989cc0850c9
branch 'master' of https://github.com/fdivitto/FabGL

Tests I tried:
GraphicalUserInterface - PS2 keyboard Test Modal Dialog typing works fine 
HardwareTest - PS2 keyboard test typing works fine
AnsiTerminal - PS2 keyboard is not detected - does not respond to any key input including F12
Altair8800 - PS2 keyboard does not respond to any key except for the F12 key

fdivitto, 2021-05-09:
Thank you for tests. Could you also test following examples?
- Others/KeyboardScanCodes
- Others/KeyboardVirtualKeys

magore, 2021-05-09:
Here is the result for Others/KeyboardVirtualKeys pressing all of the keys left to right and top down

Reset
[93m

PS/2 Keyboard Virtual Keys
Chip Revision: 1   Chip Frequency: 240 MHz
Keyboard Error!
Commands:
  1 = US Layout       2 = UK Layout       3 = DE Layout
  4 = IT Layout       5 = ES Layout
  r = Reset
  q = Scancode set 1  w = Scancode set 2
  l = Test LEDs
Various:
  h = Print This help

Use Serial Monitor to issue commands

VK_ESCAPE: 	ASCII = 0x1B		DN	[76 ]
VK_ESCAPE: 	ASCII = 0x1B		UP	[F0 76 ]
VK_F1: 	ASCII = 0x00		DN	[05 ]
VK_F1: 	ASCII = 0x00		UP	[F0 05 ]
VK_F2: 	ASCII = 0x00		DN	[06 ]
VK_F2: 	ASCII = 0x00		UP	[F0 06 ]
VK_F3: 	ASCII = 0x00		DN	[04 ]
VK_F3: 	ASCII = 0x00		UP	[F0 04 ]
VK_F4: 	ASCII = 0x00		DN	[0C ]
VK_F4: 	ASCII = 0x00		UP	[F0 0C ]
VK_F5: 	ASCII = 0x00		DN	[03 ]
VK_F5: 	ASCII = 0x00		UP	[F0 03 ]
VK_F6: 	ASCII = 0x00		DN	[0B ]
VK_F6: 	ASCII = 0x00		UP	[F0 0B ]
VK_F7: 	ASCII = 0x00		DN	[83 ]
VK_F7: 	ASCII = 0x00		UP	[F0 83 ]
VK_F8: 	ASCII = 0x00		DN	[0A ]
VK_F8: 	ASCII = 0x00		UP	[F0 0A ]
VK_F9: 	ASCII = 0x00		DN	[01 ]
VK_F9: 	ASCII = 0x00		UP	[F0 01 ]
VK_F10: 	ASCII = 0x00		DN	[09 ]
VK_F10: 	ASCII = 0x00		UP	[F0 09 ]
VK_F11: 	ASCII = 0x00		DN	[78 ]
VK_F11: 	ASCII = 0x00		UP	[F0 78 ]
VK_F12: 	ASCII = 0x00		DN	[07 ]
VK_F12: 	ASCII = 0x00		UP	[F0 07 ]
VK_PRINTSCREEN: 	ASCII = 0x00		DN	[E0 7C ]
VK_PRINTSCREEN: 	ASCII = 0x00		UP	[E0 F0 7C ]
VK_SCROLLLOCK: 	ASCII = 0x11		DN	[7E ]
VK_SCROLLLOCK: 	ASCII = 0x13		UP	[F0 7E ]
VK_DELETE: 	ASCII = 0x7F	''	DN	[E0 71 ]
VK_DELETE: 	ASCII = 0x7F	''	UP	[E0 F0 71 ]
VK_GRAVEACCENT: 	ASCII = 0x60	'`'	DN	[0E ]
VK_GRAVEACCENT: 	ASCII = 0x60	'`'	UP	[F0 0E ]
VK_1: 	ASCII = 0x31	'1'	DN	[16 ]
VK_1: 	ASCII = 0x31	'1'	UP	[F0 16 ]
VK_2: 	ASCII = 0x32	'2'	DN	[1E ]
VK_2: 	ASCII = 0x32	'2'	UP	[F0 1E ]
VK_3: 	ASCII = 0x33	'3'	DN	[26 ]
VK_3: 	ASCII = 0x33	'3'	UP	[F0 26 ]
VK_4: 	ASCII = 0x34	'4'	DN	[25 ]
VK_4: 	ASCII = 0x34	'4'	UP	[F0 25 ]
VK_5: 	ASCII = 0x35	'5'	DN	[2E ]
VK_5: 	ASCII = 0x35	'5'	UP	[F0 2E ]
VK_6: 	ASCII = 0x36	'6'	DN	[36 ]
VK_6: 	ASCII = 0x36	'6'	UP	[F0 36 ]
VK_7: 	ASCII = 0x37	'7'	DN	[3D ]
VK_7: 	ASCII = 0x37	'7'	UP	[F0 3D ]
VK_8: 	ASCII = 0x38	'8'	DN	[3E ]
VK_8: 	ASCII = 0x38	'8'	UP	[F0 3E ]
VK_9: 	ASCII = 0x39	'9'	DN	[46 ]
[... 93 lines cut ...]
VK_UP: 	ASCII = 0x00		UP	[E0 F0 75 ]
VK_END: 	ASCII = 0x00		DN	[E0 69 ]
VK_END: 	ASCII = 0x00		UP	[E0 F0 69 ]
VK_LCTRL: 	ASCII = 0x00		DN	[14 ]
VK_LCTRL: 	ASCII = 0x00		UP	[F0 14 ]
VK_LGUI: 	ASCII = 0x00		DN	[E0 1F ]
VK_LGUI: 	ASCII = 0x00		UP	[E0 F0 1F ]
VK_LALT: 	ASCII = 0x00		DN	[11 ]
VK_LALT: 	ASCII = 0x00		UP	[F0 11 ]
VK_SPACE: 	ASCII = 0x20	' '	DN	[29 ]
VK_SPACE: 	ASCII = 0x20	' '	UP	[F0 29 ]
VK_RALT: 	ASCII = 0x00		DN	[E0 11 ]
VK_RALT: 	ASCII = 0x00		UP	[E0 F0 11 ]
VK_APPLICATION: 	ASCII = 0x00		DN	[E0 2F ]
VK_APPLICATION: 	ASCII = 0x00		UP	[E0 F0 2F ]
VK_RCTRL: 	ASCII = 0x00		DN	[E0 14 ]
VK_RCTRL: 	ASCII = 0x00		UP	[E0 F0 14 ]
VK_INSERT: 	ASCII = 0x00		DN	[E0 70 ]
VK_INSERT: 	ASCII = 0x00		UP	[E0 F0 70 ]
VK_LEFT: 	ASCII = 0x00		DN	[E0 6B ]
VK_LEFT: 	ASCII = 0x00		UP	[E0 F0 6B ]
VK_DOWN: 	ASCII = 0x00		DN	[E0 72 ]
VK_DOWN: 	ASCII = 0x00		UP	[E0 F0 72 ]
VK_RIGHT: 	ASCII = 0x00		DN	[E0 74 ]
VK_RIGHT: 	ASCII = 0x00		UP	[E0 F0 74 ]

magore, 2021-05-09:
Others/KeyboardVirtualKeys reports keycodes but gets a keyboard error as before in the help menu

Reset
[93m

PS/2 Keyboard Scancodes
Chip Revision: 1   Chip Frequency: 240 MHz
Keyboard Error!
Commands:
  q = Scancode set 1  w = Scancode set 2
  l = Test LEDs       r = Reset keyboard
Various:
  h = Print This help

Use Serial Monitor to issue commands

Here is the left and right shift key
12 
F0 12 
59 
F0 59

fdivitto, 2021-05-09:
ok, with your help, I'll try to find a fix!
It seems there is communication with the keyboard, but initialisation fails. Of course I cannot replicate, even with a TTGO v1.4, so I guess it depends by the keyboard type. Maybe it needs more time to reset.
Just to start, I've just pushed a commit, could you try it?

magore, 2021-05-09:
Hi, I just pulled the updates and tried again
Things have changed - I actually got it to work about 1 out of 10 tests

Observations:
When it works:
  I get Device Id = "Unknown"  Keyboard Layout: "US"
  ( I was able to change the layout at that point )
  The leds alwasy flash just after reset (does not do this when it fails)

Between each test I erased the flash just to make sure everything was cleared out
esptool.py  --chip esp32 --port  /dev/ttyUSB0 erase_flash

This is a working test: (I was even able to change the Layout to US in this example)
PS/2 Keyboard Virtual Keys
Chip Revision: 1   Chip Frequency: 240 MHz
Device Id = "Unknown"  Keyboard Layout: "US"
Commands:
  1 = US Layout       2 = UK Layout       3 = DE Layout
  4 = IT Layout       5 = ES Layout
  r = Reset
  q = Scancode set 1  w = Scancode set 2
  l = Test LEDs
Various:
  h = Print This help

Use Serial Monitor to issue commands

magore, 2021-05-10:
FYI - the USB voltage at the board input is 4.85V. The voltage for the PS2 keyboard power pin is 4.083V. The logic signals appear to be near 3.3V according to my scope. I wonder if the PS2 power supply pin is too low (the spec calls for 5V on the power pin +/- 0.5V) Given the logic is open collector it likely will not matter as the signals are high enough. I am going to try ordering another keyboard an see what happens. I might split open a P2S extension and try boosting the supply voltage just a bit and see what happens.

fdivitto, 2021-05-10:
Actually voltage is a bit low. While logic signal can be 3.3V (IBM PS/2 tech docs say so), the power should be 5V.
Anyway, I'd like to relax a bit reset delays. I cannot commit it, but you should replace the files attached here in "src/devdrivers/keyboard.cpp" and "src/comdrivers/ps2device.cpp" folders of FabGL.
[keyboard.cpp](http://www.fabglib.org/tests/keyboard.cpp)
[ps2device.cpp](http://www.fabglib.org/tests/ps2device.cpp)

guidol70, 2021-05-10:
@fdivitto actual github master.zip does give me a compile error for RunCPM - seems to be also a PS2 source:
```
sketch\RunCPM_VGA32_v5_4_Internal_v2_3_Fix.ino.cpp.o:(.literal._Z5setupv+0x3c): undefined reference to `fabgl::PS2Controller::s_keyboard'
sketch\RunCPM_VGA32_v5_4_Internal_v2_3_Fix.ino.cpp.o:(.literal.exit._GLOBAL__sub_D_SD+0x4): undefined reference to `fabgl::PS2Controller::~PS2Controller()'
sketch\RunCPM_VGA32_v5_4_Internal_v2_3_Fix.ino.cpp.o: In function `FatVolume::chdir()':
C:\Users\guido\Documents\Arduino\libraries\SdFat\src/FatLib/FatVolume.h:67: undefined reference to `fabgl::PS2Controller::~PS2Controller()'
collect2.exe: error: ld returned 1 exit status
exit status 1
Error compiling for board ESP32 Dev Module.

```

fdivitto, 2021-05-10:
> @fdivitto actual github master.zip does give me a compile error for RunCPM - seems to be also a PS2 source:
> 

Could you try to cleanup the compiling folder? I am able to compile RunCPM with current master:
[image]

magore, 2021-05-10:
I tried the updated keyboard.cpp and ps2device.cpp files - that did not change the behaviour. I will try the power fix this afternoon and get back to you. Thanks for your help!

magore, 2021-05-10:
The power fix did not help - for the keyboard I have. Voltage now measures at 5.14V at the keyboard jack pin 4 - just a little high but in spec
For my next try I will look at the keyboard signals with my logic analyser - I see there are a few sigrok p2s protocol decoders out there which will help. I will let you know what I find.

magore, 2021-05-10:
I should also mention that I also tried uping the timeouts even more in the two files you changed to 5 seconds - and even that did not work - so something really very odd specifically with my keyboard. I will keep you updated.

magore, 2021-05-10:
It turns out that other people with this keybard have had issues with full PS2 compatibility. The model  is a Perixx PERIBOARD-409 in case you are wondering.

guidol70, 2021-05-11:
@fdivitto also got some "strange" output on/after the other F-Keys with the latest build.
on F4 through F1 it looks like assembler-code?
[image]

fdivitto, 2021-05-11:
Is this the keyboard?
https://www.amazon.it/Perixx-PERIBOARD-409-Tastiera-Laccatura-Pianoforte/dp/B07T2F9SWS/ref=asc_df_B07T2F9SWS/?tag=googshopit-21&linkCode=df0&hvadid=279818537702&hvpos=&hvnetw=g&hvrand=2600053348602412979&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=1008726&hvtargid=pla-817733137868&psc=1

It's cheap, I can afford to buy it so I try it too ;-)

magore, 2021-05-12:
FYI - another keyboard by Perixx  called the Periboard-107 works fine with the KeyboardVirtualKeys tests unlike the Periboard-409

fdivitto, 2021-05-21:
Now I also have my Perixx Periboard 409 keyboard! 
With a bit of patience and hard work with a logic analyzer, now finally it seems to work! Please install last snapshot.
FabGL has grown a lot, so there are aspects that I may not have tested, so I ask for your cooperation ...

guidol70, 2021-05-21:
> FabGL has grown a lot, so there are aspects that I may not have tested, so I ask for your cooperation ...

my working CHERRY ML-4100 keyboard keeps working ;) Thats also good :)

fretinator, 2021-05-21:
I can report that the latest MASTER code now works on all 3 of my PS/2 keyboards. Previously, 2 of them only worked with the 1.02 version. Whatever change was committed in the last week or so seems to have addressed keyboard initialization issues I was experiencing.

magore, 2021-05-26:
Hi, it is working now with the keyboard I originally reported. Thanks!
Sorry it tooks so long for me to get to your update.
Would you like me to close this request now ?

fdivitto, 2021-05-27:
Glad to hear this! Yes, please close the issue.

magore, 2021-05-27:
The issue has been resolved - perixx periboard-409 is now working

closed 2021-05-27 by magore

Miguel0101, 2021-07-12:
Does FabGL support USB keyboards?

magore, 2021-07-16:
Hi Miguel ,
     FabGL tragets, serial, PS2 (or USB keyboards with USB to PS2 
adapter plug). Many USB keyboards can support PS2 mode with an simple 
wire adapter. You can still buy USB to Keyboard/Mouse adapters on Ebay 
and Amazon - NOT to be confused with PS2 to USB adapters (wrong 
conversion direction) There are also PS2 Keyboards available.
I use a Perixx PERIBOARD-409 PS2 keyboard I got from Amazon

-- 
# Home: Mike Gore
# Mail: 405 Midwood Cres, Waterloo Ontario, N2L 5N4
# Phone: 519-884-4943
# Cell: 519-589-9303
# Email: ***@***.***
# Consulting: InfoWrite Consulting
# =========================================================
# Work: Mike Gore
# David R. Cheriton School of Computer Science, DC2609
# University of Waterloo, 200 University Ave West
# Waterloo Ontario, Canada, N2L 3G1
# Phone: 519-888-4567, x36205
# Email: ***@***.***