fdivitto/FabGL #116: [Bug] Output of "Escape-Codes"

guidol70, 2021-05-11:
@fdivitto maybe this picture will you give a hint where to search?
when I press F12 for the config-menu I often get the output
`[24~`
before the config-menu does open.
I dont know why :( With the same hardware i hadnt it yesterday, but today with the same hardware and also with another keyboard it does re-appear :(
[image]

_Originally posted by @guidol70 in https://github.com/fdivitto/FabGL/issues/115#issuecomment-838379314_

guidol70, 2021-05-11:
@fdivitto also got some "strange" output on/after the other F-Keys with the latest build.
on F4 through F1 it looks like assembler-code?
[image]

[EDIT] maybe the "new" RunCPM internal CCP 2.3 doesnt hide the codes on screen?
[EDIT2] DR-CCP does the same - will report also to RunCPM, but do see it only with the FabGL Terminal

guidol70, 2021-05-11:
@fdivitto I tested it with a FablGL master.zip from the **05.05.2021** and didnt got the output of the codes, but my next Download was the 10.05.2021 and there I got the problem :(

guidol70, 2021-05-11:
OK - I thought I often pressed F12 without seeing the Escape-Code and getting only the config-menu.
I know that I did see the Cursor-Key-Sequences in MBASIC or Wordstar.
Maybe I never pressed the keys before at this "place" because they werent needed.

Are there any Apps which need all these special keys in CP/M? I think most CP/M-Computer keyboards doesnt have all these keys.

Is there (or could there be in the future) an option to "deactivate" some keys via command.
Sure - a normal Terminal for another solution should have all keys available, but maybe it would be for some application-safety a cool/useful option to dactivate some keys.

Thanks for the link to this sequences-page. I did read some other, but they didnt compare as good as this to the codes shown on the screen ;)

closed 2021-10-04 by fdivitto

guidol70, 2021-10-12:
> About F12 it should be filtered by code when it has been intercepted.
> 

@fdivitto 
Could you please tell me how can I filter F12 for the Terminal-Output when it has been intercepted and recognized by "vkItem->vk == VirtualKey"? - because Iam did create some Toogle-Switches (F5-F8) for switching some features in my VGATextController-version.
Without the ConfDialog/Preferences I had to hardcode some settings, which I want to make soft-switchable via F-Keys.

But these F-Key ANSI outputs could interfere the Application-Outpu/-behaviour.

I do use such a code:

```
// ===============================
// F5 Toggle KeyClick
// ===============================
    if (vkItem->vk == VirtualKey::VK_F5) {
      if (!vkItem->down) {
                          if (KEYCLICK == true)
                             {KEYCLICK = false; 
                              Terminal.write("\r\nDEBUG: KEYCLICK disabled\r\n");
                             }
                             else
                              {KEYCLICK = true; 
                              Terminal.write("\r\nDEBUG: KEYCLICK enabled\r\n");
                             }                             
                             
                         } 
                                         } 
```

fdivitto, 2021-10-13:
you may set vkItem->vk = VirtualKey::NONE, on key Down and Up. Something like this:

```
  Terminal.onVirtualKeyItem = [&](VirtualKeyItem * vkItem)
  {
    if (vkItem->vk == VirtualKey::VK_F5) {
       if (!vkItem->CTRL && !vkItem->LALT && !vkItem->RALT && !vkItem->down) {
          // do something on key down
          printf("F5\n");
       }
       // always convert to NONE, on both keydown an dup
      vkItem->vk = VirtualKey::VK_NONE;
    }
  };

```